modelica/ModelicaStandardLibrary #2108: MSL uses ModelicaReference

modelica-trac-importer, 2017-01-15:
**Reported by beutlich on 11 Nov 2016 13:30 UTC**
Should the MSL have an uses annotation on ModelicaReference since there are modelica URIs inside MSL referring to it?

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2108

dietmarw, 2017-01-16:
Question is if the URI dependency in the documentation is enough to have a `uses` clause. So you could argue that technically the MSL does not depend on `Modelica Reference` to function the documentation might be targeting some dead links. This is currently not an issue since `Modelica Reference` is shipped side by side with `Modelica`. So I like to postpone this until we have a modular version of `Modelica Reference`.

beutlich, 2019-07-13:
Not a relevant issue.

closed 2019-07-13 by beutlich