modelica/ModelicaStandardLibrary #2115: Add Modelica.Icons.FunctionsPackage

modelica-trac-importer, 2017-01-15:
**Reported by beutlich on 15 Nov 2016 08:02 UTC**
Would be nice if there were a FunctionsPackage icon for packages containing functions. I do miss that icon quite a lot.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2115

modelica-trac-importer, 2017-01-15:
**Comment by dietmarw on 9 Dec 2016 10:21 UTC**
See also https://github.com/modelica/Modelica_DeviceDrivers/commit/d09a241eba7c3163ece3a085b6ee486e87ab0661 for an example.

modelica-trac-importer, 2017-01-15:
**Modified by beutlich on 9 Dec 2016 11:18 UTC**

beutlich, 2017-01-30:
See https://github.com/tbeu/ExternData/commit/3c8edfbd53075d5b72fe0e61e597224e6cf4ad65 for another example.

beutlich, 2017-01-30:
@d-hedberg If you do not have time to design a new icon, just tell me wich one you like more and I can add it for you.

beutlich, 2017-02-07:
@d-hedberg Did you have a look at this in the mean-time?

d-hedberg, 2017-02-07:
The one in https://github.com/tbeu/ExternData/commit/3c8edfbd53075d5b72fe0e61e597224e6cf4ad65 would be my preferred choice.

beutlich, 2017-02-07:
Resolved in master and maint/3.2.2.

closed 2017-02-07 by beutlich