modelica/ModelicaStandardLibrary #2117: Electrical icons not homogeneous

modelica-trac-importer, 2017-01-15:
**Reported by massimo&#46;ceraolo on 15 Nov 2016 13:54 UTC**
The "%name" field in Modelica.Electrical.MultiPhase.Sensors is too near to the sensor's circle, especially in comparison to the  position in which it is in Modelica.Electrical.Analog.Sensors.
Moreover in power sensors the "name" field is present only in the MultiPhase MSL library.

I would suggest to :
- modify the position of the field in the MultiPhase sublibrary to make it similar to the one in sublibrary Analog.Sensors
- remove "name" from Modelica.Electrical.MultiPhase.Sensors.PowerSensor

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2117

beutlich, 2017-01-16:
Point to @ceraolo.

christiankral, 2017-03-29:
This issue is taken care of in #2215

closed 2017-04-18 by beutlich