modelica/ModelicaStandardLibrary #2120: Blocks.Tables lack of functionality / functions

modelica-trac-importer, 2017-01-15:
**Reported by ahaumer on 23 Nov 2016 13:34 UTC**
The block CombiTimeTable has some unique functionality reps. functions:
* Parameter extrapolation defines what happens if time goes beyond the last point in time defined in the table.
* The user is able to retrieve the first and the last point in time defined in the table, even when the table is read from file.

It would be a great advantage if the same functionality would be available at least in the blocks CombiTable1D and CombiTable1Ds which do pretty much the same as CombiTimeTable except that the independent variable is an input instead of time. The other parameters are pretty much the same. 
The enhancement could be done in a backwards compatible way rather easily: The additional functions do not harm at all, and the parameter extrapolation could be set by default to Extrapolation.LastTwoPoints.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2120

modelica-trac-importer, 2017-01-15:
**Comment by ahaumer on 23 Nov 2016 20:12 UTC**
In general for the CombiTable1D(s) not every type of extrapolation makes sense:
* HoldLastPoint yes (new choice)
* LastTwoPoints yes (current behaviour)
* Periodic no (see remark below)
* NoExtrapolation yes (new choice)
"Periodic" only makes sense in an application where the input is either strict monotonically rising or falling like time for the CombiTimeTable.

Anyway, the functions that return minimum and maximum of the table abscissa would be extremly useful.

modelica-trac-importer, 2017-01-15:
**Comment by ahaumer on 23 Nov 2016 20:48 UTC**
Maybe I'm wrong, but periodic extrapolation might make sense:
Think about defining the inclination of a round track defined with respect to the position on the track. If the vehicle leaves the end of the track it arrives at the beginning of the track. If the vehicle reverses direction of movement, it goes backwards over the course. 
I'm not sure whether this behaviour makes sense in general.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 24 Nov 2016 11:46 UTC**
milestone:Design91 was the MDM in Regensburg. So I switched to \*undeciced\* for now.

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 25 Nov 2016 10:34 UTC**
See also #477 for the extrapolation options.

beutlich, 2017-02-25:
@AHaumer Resolved in master.

christiankral, 2017-02-26:
Periodic continuation would also make sense for
- ``Modelica.Blocks.Sources.BooleanTable``
- ``Modelica.Blocks.Sources.IntegerTable``

beutlich, 2017-02-26:
Currently periodic extrapolation only is available for the CombiTables based on external objects BooleanTable, IntegerTable and TimeTable do no have the extrapolation feature at all. Your workaround would be to pass a Real signal from a CombiTable to Integer or Boolean. Please open a new issue for these features on the built-in tables as it is a separate topic.

closed 2017-02-26 by beutlich