modelica/ModelicaStandardLibrary #2121: C_default not defined in Modelica.Media.Interfaces.PartialMedium

modelica-trac-importer, 2017-01-15:
**Reported by rubenbaetens on 24 Nov 2016 09:41 UTC**
Modelica.Media.Interfaces.PartialMedium defines T_default, h_default, X_default and p_default, but no C_default. It would be better to add C_default = fill(0, Medium.nC) to the MSL so that all is consistent. As such, users can use C_default as the default start value and the default concentration for Boundary_pT, and can reassign this value in the medium.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2121

beutlich, 2017-02-07:
Pointing to @rubenbaetens.

beutlich, 2017-04-04:
@hubertus65 Any feedback on this enhancement?

beutlich, 2017-10-14:
@rubenbaetens @casella Since this is [Hacktoberfest](https://twitter.com/hashtag/Hacktoberfest) and the library officers did not give any feedback I created #2364 now.

PR #2364 (2018-01-21): Add missing C_default to PartialMedium
Close #2121.

@rubenbaetens FYI
closed 2018-01-21 by hubertus65