modelica/ModelicaStandardLibrary #2131: Ugly icon for Modelica.SIunits

modelica-trac-importer, 2017-01-15:
**Reported by dag on 12 Dec 2016 14:05 UTC**
Modleica.SIunits has an icon which uses the unit "kg" (kilogram). That is ok. The problem is that the letters "kg" are drawn using polygons instead of text. Proposal: use Text(...) annotation instead.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2131

tbeu, 2017-01-31:
@DagBruck Text in icons is problematic since,
* tools behave different on text size and font
* nice looking text in the icon view does not mean it renderes nicely in the package browser. See https://github.com/modelica-3rdparty/msgpack-modelica/commit/961aaf6ac0129b2fdb83c0f5bd0e0446dfa699dc for an example.

henrikt-ma, 2017-02-01:
Given how much time we've spent recently on how to render ellipses,
https://trac.modelica.org/Modelica/ticket/2045
I think it would make sense to then consider the more important question of rendering of text.  That is, arguing that tools behave differently when rendering text should not be an argument against having text in icons.

tbeu, 2017-02-01:
@henrikt-ma You might be interested to follow the modelica/Modelica_DeviceDrivers#186 thread.

d-hedberg, 2017-02-07:
Here's a proposal for a new SIunits package icon. Feel free to use it or modify it to your liking.

```
package SIunits
  extends Modelica.Icons.Package;
  annotation(Icon(graphics = {Polygon(visible = true, fillColor = {128, 128, 128}, pattern = LinePattern.None, fillPattern = FillPattern.Solid, points = {{-80, -40}, {-80, -40}, {-55, 50}, {-52.5, 62.5}, {-65, 60}, {-65, 65}, {-35, 77.5}, {-32.5, 60}, {-50, 0}, {-50, 0}, {-30, 15}, {-20, 27.5}, {-32.5, 27.5}, {-32.5, 27.5}, {-32.5, 32.5}, {-32.5, 32.5}, {2.5, 32.5}, {2.5, 32.5}, {2.5, 27.5}, {2.5, 27.5}, {-7.5, 27.5}, {-30, 7.5}, {-30, 7.5}, {-25, -25}, {-17.5, -28.75}, {-10, -25}, {-5, -26.25}, {-5, -32.5}, {-16.25, -41.25}, {-31.25, -43.75}, {-40, -33.75}, {-45, -5}, {-45, -5}, {-52.5, -10}, {-52.5, -10}, {-60, -40}, {-60, -40}}, smooth = Smooth.Bezier), Polygon(visible = true, fillColor = {128, 128, 128}, pattern = LinePattern.None, fillPattern = FillPattern.Solid, points = {{87.5, 30}, {62.5, 30}, {62.5, 30}, {55, 33.75}, {36.25, 35}, {16.25, 25}, {7.5, 6.25}, {11.25, -7.5}, {22.5, -12.5}, {22.5, -12.5}, {6.25, -22.5}, {6.25, -35}, {16.25, -38.75}, {16.25, -38.75}, {21.25, -41.25}, {21.25, -41.25}, {45, -48.75}, {47.5, -61.25}, {32.5, -70}, {12.5, -65}, {7.5, -51.25}, {21.25, -41.25}, {21.25, -41.25}, {16.25, -38.75}, {16.25, -38.75}, {6.25, -41.25}, {-6.25, -50}, {-3.75, -68.75}, {30, -76.25}, {65, -62.5}, {63.75, -35}, {27.5, -26.25}, {22.5, -20}, {27.5, -15}, {27.5, -15}, {30, -7.5}, {30, -7.5}, {27.5, -2.5}, {28.75, 11.25}, {36.25, 27.5}, {47.5, 30}, {53.75, 22.5}, {51.25, 8.75}, {45, -6.25}, {35, -11.25}, {30, -7.5}, {30, -7.5}, {27.5, -15}, {27.5, -15}, {43.75, -16.25}, {65, -6.25}, {72.5, 10}, {70, 20}, {70, 20}, {80, 20}}, smooth = Smooth.Bezier)}));
end SIunits;
```

sjoelund, 2017-02-07:
@henrikt-ma I think the problem is not just how to render text (all tools I know how deal with absolute font sizes very differently, but render fine with auto-sized text fitting a bounding box), but also specifying when to render text. Most tools tend to remove rendering of text in the package browser since text on ports, etc. tends to be small and irrelevant. This makes using polygons for package icons a reasonable workaround until it's specified in a good way when to render text...

@d-hedberg I think that icon is a huge improvement over the current one in MSL. It's 1652 characters vs. ~1067 for the existing one, which is the only drawback.

dietmarw, 2017-02-07:
1.65 kB vs 1.06 kB .. I think we can cope with this "space wasting" improvement :smiley:

beutlich, 2017-02-07:
@d-hedberg Thanks for the improved Icon graphics.

Resolved in master and maint/3.2.2 branches.

closed 2017-02-07 by beutlich