modelica/ModelicaStandardLibrary #2140: initial condition of SlewRateLimiter

modelica-trac-importer, 2017-01-15:
**Reported by ahaumer on 26 Dec 2016 09:52 UTC**
The initial condition of the SlewRateLimiter is wrong in one special case: apply a step at time=0 at the input, I'd expect a rise starting at the offset of the step, but the block takes the input after the step as initial value. Maybe let the user define the initial output is a better solution.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2140

modelica-trac-importer, 2017-01-15:
**Comment by ahaumer on 26 Dec 2016 10:07 UTC**
I see: If I change the implementation of Step to

```
y = offset + (if time <= startTime then 0 else height);
```
(note the "less equal" instead of "less than"), the initial condition of the slew rate limiter works fine.
Maybe we should adapt the implementation to the Step block?

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 2 Jan 2017 15:06 UTC**
Replying to [comment:1 ahaumer]:
> I see: If I change the implementation of Step to
> 
> ```
> y = offset + (if time <= startTime then 0 else height);
> ```
> (note the "less equal" instead of "less than"), the initial condition of the slew rate limiter works fine.

That isn't good: `time<=startTime` generates state events instead of time events. Not just initially - but everywhere, and in every model.

To me it makes more sense that SlewRateLimiter would have the possibility to set the starting conditions. This goes together with the general idea that it is desirable to be able to disable initial equations (such as `y=u`).

AHaumer, 2017-05-14:
resolved in master, added an example.
Additionally changed the type of parameters "Rising" and "Falling" to Real, otherwise an application using physical parameters for "Rising" and "Falling" (as shown in the example: velocity resp. acceleration) would rise an automatic unit deduction failure.

beutlich, 2017-08-05:
The new example model misses the experiment.StopTime annotation.

commit d632d02f (2017-05-14): resolves #2140
commit 7a6c95eb (2017-12-03): resolves #2140
closed 2017-12-03 by AHaumer