modelica/ModelicaStandardLibrary #2146: Missing initial Conditions for Modelica.Electrical.Analog.Semiconductors.NPN

modelica-trac-importer, 2017-01-15:
**Reported by jfrenkel on 6 Jan 2017 12:20 UTC**
The attached model evoke in different Modelica compilers missing initial conditions for the model Modelica.Electrical.Analog.Semiconductors.NPN. 

If `UIC` is `false` there are two missing initial conditions. If `UIC` is `true` then one missing initial condition is reported.

This should be fixed to avoid confusion by the users.

A possible solution is to add an parameter to initialize the Variable `vbe`

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2146

modelica-trac-importer, 2017-01-15:
**Comment by beutlich on 6 Jan 2017 12:31 UTC**
Please do also add a description to parameter `UIC`.

kristinmajetta, 2018-05-03:
Dymola does not report missing initial conditions for that model in both cases (UIC=true and UIC=false).

Added a description string to UIC in SHA-1: ae6372e926224a5aaeb9893e9c19e504fd481091

beutlich, 2018-05-04:
Dymola 2018 FD01 reports in pedantic mode for UIC=false:
```
The initial conditions are not fully specified.
Dymola has selected default initial conditions.

Assuming fixed default start value for the continuous states:
npn1.C.v(start = 0.0)
npn1.vbe(start = 0.0)

```
and for UIC=true:
```
The initial conditions are not fully specified.
Dymola has selected default initial conditions.

Assuming fixed default start value for the continuous states:
npn1.vbe(start = 0.0)
```
but there is no clean way to specifiy the fixed start / initial value for protected state variable vbe.

kristinmajetta, 2018-05-09:
The initial condiction handling for capacitances need a generally suitable concept. This concerns other models containing capacitances besides the NPN model. We do not prefer a " quick and dirty" solution for this particular ticket. We postpone the ticket until we developed the concept for handlig the initial conditions.

dietmarw, 2018-05-09:
There would be a simple fix to at least allow the user to get rid of this annoying warning. Make `vbe` unprotected. I don't see this as hack but rather fix.

beutlich, 2019-10-23:
> There would be a simple fix to at least allow the user to get rid of this annoying warning. Make `vbe` unprotected. I don't see this as hack but rather fix.

The short-term issue was resolved by 5eaf686e024d61533d1095b421fbaeddb084f385, aa6b3d0b66c5bad8d15d86981b691ec34c941618 and 68f6be7ed9dd65c8fffe7e4e0d85c58e69f37640 already for MSL v3.2.3.

> We postpone the ticket until we developed the concept for handlig the initial conditions.

The long-term issue will be considered in #3168. It is unlikely that it will be resolved for MSL v4.0.0.

closed 2019-10-23 by beutlich

christiankral, 2019-10-29:
I have some more comments on this issue here as #3168 seems (from the title) not the be right place to put them: 

1. The NPN model contains the following parameters:

```modelica
parameter SI.Voltage IC=0 "Initial value" annotation(Dialog(enable=UIC));
parameter Boolean UIC = false "Decision if initial value should be used";
```
The parameter description string of `IC` states `Initial value` but it is not clear which quantity this value refers to. Reading the code the initial value `IC` is the potential of the collector, `C.v`.

2. I find it strange the the potential of pin `C` instead of a potential difference (voltage) is set by the initial condition:

```modelica
initial equation 
  if UIC then
    C.v = IC;
  end if;
```

Consequently the following circuit is illegal when `UIC` is true, as the potential of the collector is fixed by `constantVoltage2` already. 

[image]

It does absolutely not make sense to me that such circuit shall not be allowed! This is not only a shortcoming of the model, this is a **bug** on my opinion.

3. The three equations related with the transistor currents

```modelica
  C.i = (ibe - ibc)*qbk - ibc/Br - cbc*der(vbc) + Ccs*der(C.v);
  B.i = ibe/Bf + ibc/Br + cbc*der(vbc) + cbe*der(vbe);
  E.i = -B.i - C.i + Ccs*der(C.v);
``` 

show the expression `der(C.v)` in the first and third line. This does not make sense to me. The parameter description of the capacitance `Ccs` reads

```modelica
parameter SI.Capacitance Ccs=1e-12 "Collector-substrate(ground) cap.";
```

so obviously the corresponding capacitor `Ccs` is connected between the collector and the substrate which is assumed to have ground potential. It, however, does not make sense to have implicit grounding in the transistor model. Instead a substrate connector shall be added if the effect of the collector substrate capacitance shall be taken into account. 

**Note:** When reading [Vlach1983] I could neither find the capacitance `Ccs` nor the substrate of the bipolar transistor model, but I may have overseen the corresponding equations and text in the book.

dietmarw, 2019-10-29:
I agree with @christiankral  that the implicit earthing of the component makes no sense unless it has an additional connector that can be set to a certain potential. I suggest to open a new ticket for this.

christiankral, 2019-10-29:
The included test model is available as [Transistor2.mo](https://gist.github.com/christiankral/d226a3657583cd434264cedab635832c)

beutlich, 2019-10-29:
It is also odd that UIC/IC are only present in NPN, but not in HeatingNPN. I suggest to remove it then from #3178 where I explicitly added it for backward compatibility yesterday.

christiankral, 2019-10-29:
> It is also odd that UIC/IC are only present in NPN, but not in HeatingNPN. I suggest to remove it then from #3178 where I explicitly added it for backward compatibility yesterday.

I fully agree. Even the NPN and PNP model are not implemented in the same way as only the NPN includes an initial condition. Yet, I think we shall just go ahead with #3178 and merge and then create a different PR handling  

- the initial conditions 
- the substrate connector

beutlich, 2019-10-29:
OK, #3178 is merged with only NPN/PNP left to consider.