modelica/ModelicaStandardLibrary #2147: Duplicated function pow in Modelica.Electrical.Analog.Semiconductors

modelica-trac-importer, 2017-01-15:
**Reported by beutlich on 6 Jan 2017 12:47 UTC**
Remove duplicated functions Modelica.Electrical.Analog.Semiconductors.NPN.pow and Modelica.Electrical.Analog.Semiconductors.PNP.pow in favor of Modelica.Electrical.Analog.Semiconductors.pow.

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2147

modelica-trac-importer, 2017-01-15:
**Comment by hansolsson on 10 Jan 2017 12:26 UTC**
Replying to [ticket:2147 beutlich]:
> Remove duplicated functions Modelica.Electrical.Analog.Semiconductors.NPN.pow and Modelica.Electrical.Analog.Semiconductors.PNP.pow in favor of Modelica.Electrical.Analog.Semiconductors.pow.

Having a function "pow" there looks odd to me.

I would prefer either use `x^y` directly or have it some-place neutral, like Modelica.Math.pow, or possibly just keep them as local helper functions - since that makes the individual semiconductors models self-contained.

tbeu, 2017-01-31:
@DagBruck @HansOlsson The pow helper functions were introduced by 4461da7d2969d2b22b10d3cb8b8eb73cbfead78b 13 years ago, probably because Dymola required it. Can we remove them safely now? They are used in expressions like `Capcjc = smooth(1,(if (vbc/Phic > 0) then Cjc*(1 + Mc*vbc/Phic) else Cjc*pow(1 - vbc/Phic, -Mc)))`.

HansOlsson, 2017-02-03:
Using `x^y` directly is supported in Dymola, since a long time so you can just use it. 
Are you sure it was because of an tool issue - I had thought it was a matter of readability, since you easily get lost in ` Capcjc = smooth(1,(if (vbc/Phic > 0) then Cjc*(1 + Mc*vbc/Phic) else Cjc*(1 - vbc/Phic)^(-Mc))) `.

tbeu, 2017-02-03:
No, I only guessed that it was a tool issue of the Dymola symbolic engine telling from the history and the comment given by https://github.com/modelica/Modelica/blob/master/Modelica/Electrical/Analog/Semiconductors.mo#L1571.

beutlich, 2017-04-18:
Resolved in master.

closed 2017-04-18 by beutlich