modelica/ModelicaStandardLibrary #2148: Simplify Initialization of Modelica.Thermal.FluidHeatFlow.Interfaces.Partials.TwoPort

FrenkelJ, 2017-01-23:
The initialization of extended models from Modelica.Thermal.FluidHeatFlow.Interfaces.Partials.TwoPort would be much easier if the variable h get as modification of the start attribute
`
Modelica.SIunits.SpecificEnthalpy h(start=medium.cp*T0) "Medium's specific enthalpy";
`

Ad least the example Modelica.Thermal.FluidHeatFlow.Examples.PumpAndValve would benefit from the additional information.

casella, 2017-01-31:
@FrenkelJ can you open a ticket on the MSL issue tracker?

sjoelund, 2017-01-31:
This is the MSL issue tracker. trac.modelica.org is read-only for MSL components.

beutlich, 2017-01-31:
@casella Both the MA Webmaster, @AHaumer, @MartinOtter and me sent several mails that the MSL is now developed on GitHub, including issues. It also was tweeted by https://twitter.com/modelica/status/820906213806374912 two weeks ago.

casella, 2017-01-31:
Sorry folks, the management of the Modelica Conference papers obviously hit hard on me. Please be patient with me and allow me to catch up again soon.

That said, I agree with @FrenkelJ proposal

beutlich, 2017-01-31:
Also cherry-picked to maint/3.2.2 branch.

PR #2154 (2017-01-31): close #2148
close #2148
Add modification of h.start to simplify initialization
commit 7c5aefce (2017-01-31): close #2148
Add modification of h.start to simplify initialization
commit 882d0087 (2017-01-31): close #2148
Add modification of h.start to simplify initialization
closed 2017-01-31 by beutlich