modelica/ModelicaStandardLibrary #2152: Remove deprecated Text.lineColor attribute in graphical annotations

beutlich, 2017-01-30:
Text.lineColor (as all other contents of FilledShape) were deprecated in MLS v3.3 and should be replaced by Text.textColor in an MSL based on MLS >= v3.3.

@HansOlsson How easy can you figure out all the occurences of deprecated FilledShape attrubutes on Text in MSL? There are plenty of them ...

HansOlsson, 2017-02-03:
@beutlich We already have logic in Dymola to allow users correct some other annotations (flipped Bitmaps and misuse of LinePattern.None for connection-lines) and it was straightforward to re-use that logic to replace lineColor by textColor and remove the other ones (there are some uses of lineThickness, fillColor, fillPattern, and even pattern).

Shall I make a pull request based on that - or just removing the meaningless attributes, and keep lineColor?Note that some tools do not yet generate the Modelica 3.3 annotations for textColor - since lineColor is more compatible.

BTW: Since I extended the previous routine it modified one connection line in Modelica.Magnetic.FluxTubes.Examples.MovingCoilActuator.Components.PermeanceActuator - I will check if this still happens, and in that case open a separate issue on that. (I could understand that you sometimes want invisible connection lines - but not that one.)
--

beutlich, 2017-02-03:
@HansOlsson A PR (=pull request) against master branch is highly appreciated. For now, Keeping lineColor and removing the silly Text attributes should cause no harm. Thank you.

HansOlsson, 2017-07-10:
As far as I understand the merged pull-requests closed this issue.

tbeu, 2017-07-10:
No, this is the task for the MSL based on Modelica > 3.2. As long as we do not know, what the master branch will be based on for the next release, we have to keep it open (and use textColor attribute once we know that master can use Modelica 3.3 features).

PR #3014 (2019-07-01): Prefer Text.textColor attribute (and do no longer use deprecated Text.lineColor)
Here's how I did it in Notepad++

- Find what: `(Text\s*\([^\).]*)lineColor`
- Replace with: `(\1)textColor`

[image]

Closes #2152.
closed 2019-07-01 by beutlich