modelica/ModelicaStandardLibrary #2157: ComplexBlocks.Sources.LogFrequencySweep is not complex

JoeRiel, 2017-02-06:
Why is the LogFrequencySweep in the ComplexBlocks package?  Its output is real. Shouldn't it be in Blocks.Sources?

tbeu, 2017-02-07:
This requires a conversion script.

beutlich, 2017-02-07:
This requires a conversion script.

See #1850 for the corresponding ticket that added this block.

AHaumer, 2017-03-05:
The block is mainly used in the ComplexBlock - Examples.

AHaumer, 2018-01-03:
It should go rather to a Utilities package of ComplexBlocks ...

AHaumer, 2018-05-18:
As far as I remmebr, the block can't be moved now due to backwards compatibility,

beutlich, 2018-05-18:
> As far as I remmebr, the block can't be moved now due to backwards compatibility,

Exactly. The milestone is set to MSL_next-MAJOR-version for that reason.

christiankral, 2019-07-02:
I will move `ComplexBlocks.Sources.LogFrequencySweep` to `Blocks.Source`, as it is a signal source: this is where it belongs to.

beutlich, 2019-07-02:
Havea look at https://github.com/modelica/ModelicaStandardLibrary/pull/2158/files. Might save you some typing.

PR #3020 (2019-07-05): Move LogFrequencySweep from ComplexBlocks.Sources to Blocks.Sources
Closes #2157
closed 2019-07-05 by beutlich