modelica/ModelicaStandardLibrary #2161: CONTRIBUTING.md and other special github files

thorade, 2017-02-08:
github knows a couple of [special files](https://help.github.com/categories/building-a-strong-community/)
- [LICENSE.md](https://help.github.com/articles/adding-a-license-to-a-repository/) (for the license text)
- [CONTRIBUTING.md](https://help.github.com/articles/setting-guidelines-for-repository-contributors/) (for contributing guidelines)
- [ISSUE_TEMPLATE](https://help.github.com/articles/creating-an-issue-template-for-your-repository/)
- [PULL_REQUEST_TEMPLATE](https://help.github.com/articles/creating-a-pull-request-template-for-your-repository/)
- [CODE_OF_CONDUCT.md](https://help.github.com/articles/adding-a-code-of-conduct-to-your-project/)
- .gitattributes (really from git, but github places [some things](https://github.com/github/linguist#overrides) here, too)

These files, especially the CONTRIBUTING.md, can make life easier for first-time contributors and help with an efficient workflow, describing how to properly fork a repository, create issue/topic branches, cross-reference issues, submit a pull request and so on.
Another possibly helpful feature are [Pull Request reviews](https://help.github.com/articles/about-pull-request-reviews/) in combination with [protected branches](https://help.github.com/articles/about-protected-branches/).

Do you want me to submit a PR for an inital version of a CONTRIBUTING.md file?

dietmarw, 2017-02-08:
Good idea but I would wait with it until people from the MA have started getting used to the new system first. But please go ahead with the PR on CONTRIBUTING.md. So can we discuss this there. 
BTW, `.gitattributes` is already there. So improvements can be added of course (but make that a separate pull-request.

HansOlsson, 2017-02-24:
For ISSUE_Template the following would have helped me:

1. Make the issue-text brief, use attachments for larger documents.
2. If you know exactly what to correct, just make a pull-request directly.

Note that the first item indicates that we should not make an overly large template. 
Just something simple as:
```
The model <full model name> <describe issue>
<Describe the problem in detail - if the issue is incorrect graphics or simulation behavior, first check that it is not a tool-issue and then describe the observed and expected behavior. Use attachments for larger texts and other documents.>
<If you know how to correct the issue, you can directly create a pull request instead with the same information.>
```
But we should also ask others.

beutlich, 2017-02-26:
For the ISSUE_Template also the MSL version having the issue is relevant. There was such a version flag in Trac that no longer is available at GitHub issues.

thorade, 2017-06-15:
Also see this github blog post:
https://github.com/blog/2380-new-community-tools
And this page:
https://github.com/modelica/Modelica/community
It shows a checklist for some of the files mentioned above.

PR #2187 (2018-03-23): Issue2161 contributing
An initial version for Contributing guidelines.
This would close #2161 for now.
closed 2018-03-23 by dietmarw