modelica/ModelicaStandardLibrary #2162: Diode2 has line from anode to cathod

JoeRiel, 2017-02-08:
The icon for Modelica.Electrical.Analog.Semiconductors.Diode2 has a line from the negative terminal to the positive terminal.  It would be better if this were broken into two lines, one from the negative terminal to the cathode bar, another from the positive terminal to anode body.  The problem with the line through the body is that it is indistinguishable from a direct connection between the nodes, which can happen accidentally while creating a schematic.

beutlich, 2017-02-09:
Is this only specific for Diode2 or also Diode and ZDiode?

ceraolo, 2017-03-06:
This line is present also in the voltage sources (except ConstantVoltage, which in turn is being changed, see #2186).   **The real issue is that in IEC standards these symbols do have the connection line!**

We have two options:
1) use quasi-IEC symbols we are currently using and keep the line connecting the two terminals from inside the component 
2) remove the connecting line, as suggested by this ticket, and move away from the standard. 

I don't like any of these two. One of the beauty of MSL is that electric circuits can be directly exported  in reports and papers, and, after exporting, they are very similar to how they are when drawn from scratch. This is in favour of option 1). But as a teacher I know very well that the connecting line through the terminals, passing inside the components is error prone (this is an argument in favour of option 2)

As a third way I can envisage using a different colour for the line-through terminals inside components, for instance black. 
Converting all electrical components into black (and leaving in blue only terminals and connecting lines) seems overkill. Using black only for the pin-to-pin inside connections would result in rather ugly icons, but probably constitutes a better option than 1) and 2).

If we must stick to options 1) or  2) I would prefer 1) not as the best, but as the "worser" (I don't know if this is good English)

kristinmajetta, 2017-04-04:
Since option 1) is the closest to the IEC standard, I am also in favour of option 1)

kristinmajetta, 2017-06-28:
The symbol of the diode models were created according to the IEC standard, where the line between the two pins is present. One possibility would be to introduce different symbols according to different standards (e.g. the american diode has no line) and the user can switch between the standards (and symbols) via a parameter or something. As long as that is not possible, the diode symbols are left as they are right now - WITH the line.

closed 2017-06-28 by kristinmajetta

dietmarw, 2017-06-28:
Just as a note. Different appearance depending on a parameter is already possible and used by various libraries. Have a look at the `visible` annotation.

HansOlsson, 2017-06-28:
Using the `visible` annotation seems good, but:
If we are going to add different variants (e.g. American ones) - I don't want us to start adding parameters on each model, but have a general setting automatically applicable to everything.

Normally this would be seen as part of internationalization - together with translation of descriptions etc (see https://trac.modelica.org/Modelica/ticket/302 ).