modelica/ModelicaStandardLibrary #2164: Replace Blocks.Logical

beutlich, 2017-02-11:
Package Modelica.Blocks.MathBoolean contains the following comment since 2010, that is MSL v3.2.
> Package MathBoolean is a new design that shall replace in the future the Logical package.

This is a reminder for it.

HansOlsson, 2019-02-13:
I agree that we shouldn't have two such similar packages.
As I see it there are three parts of this:

- Different package name. MathBoolean instead of Logical. I agree that it MathBoolean works better alongside MathInteger, but Logical seems more logical.
- Smaller size. Ok I guess.
- Multiple inputs. Overall I find that it works, but for Xor it is not obvious - as there is another way of generalizing Xor to multiple inputs: parity check, i.e. sum modulo 2; and we don't have that. To me it would be clearer to have three blocks: 2-input Xor, n-input Parity check and n-input OneTrue.

Oh, and for the tools I realize that the connectorSizing adds an interesting issue for the conversions.

beutlich, 2019-02-13:
@MartinOtter What is the current state of replacing Modelica.Blocks.Logical by Modelica.Blocks.MathBoolean? Given the comparison table, only the Modelica.Blocks.MathBoolean.OnDelay block is not represented in Modelica.Blocks.Logical.

|Logical | MathBoolean|
|--------|------------|
|And |And |
|Or |Or |
|Xor |Xor |
|Nor |Nor |
|Nand |Nand |
|Not |Not |
|Pre | |
|Edge |RisingEdge |
|FallingEdge |FallingEdge |
|Change |ChangingEdge |
|GreaterThreshold | |
|GreaterEqualThreshold | |
|LessThreshold | |
|LessEqualThreshold | |
|Greater | |
|GreaterEqual | |
|Less | |
|LessEqual | |
|ZeroCrossing | |
|LogicalSwitch |MultiSwitch |
|Switch | |
|Hysteresis | |
|OnOffController | |
|TriggeredTrapezoid | |
|Timer | |
|RSFlipFlop | |
|TerminateSimulation | |
| |OnDelay |

HansOlsson, 2019-02-13:
> Given the comparison table, only the Modelica.Blocks.MathBoolean.OnDelay block is not represented in Modelica.Blocks.Logical.

I tried to verify the table. I believe that MultiSwitch isn't the obvious generalization of LogicalSwitch - but something completely different. For the future I believe it might be good if we have vector-variants of both LogicalSwitch and Switch - with integer input.

beutlich, 2019-02-13:
Thanks. Edited above.

beutlich, 2019-02-13:
See also #2679 and the proposal/question from @tobolar 

> I understand the application area of the blocks could be different, but shouldn't it be better to e.g. extend Logical.Edge from MathBoolean.RisingEdge?

tobolar, 2020-11-27:
There are two more classes which could be replaced as well, both being in `Modelica.Blocks.Math`:
- Edge
- BooleanChange