modelica/ModelicaStandardLibrary #2165: Transition from Suberversion to Git

beutlich, 2017-02-12:
After the transition to GitHub
* the Modelica.UsersGuide.ReleaseNotes.VersionManagement needs to be updated to reflect the current MSL hosting and workflows,
* the revisionId annotation should be removed.

dietmarw, 2017-02-13:
There is no need to remove `revisionId` since that can also be facilitated by use of the [`export-subst` git attribute](https://git-scm.com/docs/gitattributes) when doing a [git-archive](https://git-scm.com/docs/git-archive) for a release (or especially pre-release).

beutlich, 2017-02-15:
See [doc.modelica.org](http://doc.modelica.org/help/Modelica_UsersGuide_ReleaseNotes.html#Modelica.UsersGuide.ReleaseNotes.VersionManagement) for how it Looks today.

beutlich, 2018-03-21:
Proposal: Add link to a (future) wiki/Contributing.md from Modelica.UsersGuide.ReleaseNotes.VersionManagement