modelica/ModelicaStandardLibrary #2168: Fix documentation of ComplexMath.Asin, Acos and Log

beutlich, 2017-02-13:
The input constraints do not hold for complex numbers and seem to have been copy/pasted from the corresponding blocks of Blocks.Math.

For Asin/Acos:
```html
The absolute values of the elements of the input <code>u</code> need to
be less or equal to one (<b>abs</b>( u ) <= 1).
Otherwise an error occurs.
```
For Log:
```html
An error occurs if the elements of the input <code>u</code> are
zero or negative.
```

AHaumer, 2017-03-05:
Agree for Asin and Acos, but how should Complex u be negative? abs(u) is always >= 0, ok for abs(u)=0 we have a problem.

HansOlsson, 2017-03-06:
> Agree for Asin and Acos, but how should Complex u be negative? abs(u) is always >= 0, ok for abs(u)=0 we have a problem.

Well, -1 seems negative, but `Modelica.ComplexMath.log(Complex(-1))` is ok - and thus it seems it should be:
`An error occurs if the elements of the input <code>u</code> are zero.`

Similarly `Modelica.ComplexMath.asin(Complex(2,0)` actually works - so we don't need any restriction on those.

However, there is another issue (needs a separate ticket for Modelica.Blocks.Math; will open it) - it is a single input and thus I would expect:
Log: `An error occurs if the input <code>u</code> is zero.`
Asin/Acos: _This space intentionally left blank_

It is likely that the error in Modelica.Blocks.Math occurred due to Modelica 1->2 change when blocks changed from having vectors to scalars as inputs.

PR #2265 (2017-06-15): Improve documentation of Blocks
Resolve #2203  by describing scalars as scalars and not as vectors. 
Resolve #2168 by not limiting complex Acos and Asin, and by only restricting Log to non-zero values.
closed 2017-06-15 by MartinOtter