modelica/ModelicaStandardLibrary #2171: Line force length assertion in Mechanics.MultiBody.Forces

tbeu, 2017-02-16:
There is a line force length assertion `length > s_small` in MultiBody.Forces.Spring (coming from the component of type LineForceWithMass). However, this assertion is missing in MultiBody.Forces.Damper, MultiBody.Forces.SpringDamperParallel and MultiBody.Forces.SpringDamperSeries. I propose to also add the line force length assertion `Modelica.Math.Vectors.length(r_rel_a) > s_small` to MultiBody.Interfaces.PartialLineForce to have it consistent.

HansOlsson, 2017-02-17:
Overall the inheritance seems odd in the MultiBody-library, causing the code to be duplicated - and sometimes missing.

It seems as though PartialLineForce could inherit from PartialTwoFrames (common code for cardinality-assert). And then LineForceWithMass and LineForceWithTwoMasses have something in common with PartialTwoFrames - not only the length check, but also fixedRotationAtFrame_*.

beutlich, 2017-03-28:
> And then LineForceWithMass and LineForceWithTwoMasses have something in common with PartialTwoFrames -

I guess it should read: ... have something in common with PartialLineForce

beutlich, 2017-04-03:
Resolved in master branch by 519c2c581fd2f17c859dffda2cd0f921b2701b9a and f56af70e7d194c227bcf70595363806f0b91a1e0.

closed 2017-04-03 by beutlich