modelica/ModelicaStandardLibrary #2172: Implementation of visualization of phasor diagrams

christiankral, 2017-02-16:
I recently developed the idea of using ``Modelica.Mechanics.MultiBody.Visualizers`` to draw phasor diagrams of quasi stationary electric circuits. For this purpose I am planning to extend the Interface definitions of ``Modelica.Electrical.QuasiStationary.SinglePhase.Interfaces.TwoPin`` by visual elements of ``Modelica.Mechanics.MultiBody``. The phasors shall then be displayed as 2D (without considering z coordinates) vectors in space (with units meter). 

I am planning to create a new development branch based on master. The new implemented proposal  could then be integrated back into MSL master.

beutlich, 2017-02-16:
@christiankral You set milestone to MSL_next-MINOR-version (probably MSL v3.2.3) but plan to branch of from master. I'd recommend to branch of from maint/3.2.2 branch instead (or set milestone to MSL_next-MAJOR-version (probably MSL v3.3, v3.4 or v4.0) alternatively).

christiankral, 2017-02-16:
@beutlich OK I will branch it from maint/3.2.2 in order to to consider it for the MSL_next-MINOR-version.

beutlich, 2017-02-16:
@christiankral I removed my above comment, since I was wrong. The next MSL release will be built from master. Since there is/was no decision if the next MSL release will be a backward-compatible release (MSL_next-MINOR-version) or a compatibility-breaking release (MSL_next-MAJOR-version), the master must not have any compatibility-breaking commits which is the reason why so many issues/patches for MSL_next-MAJOR-version are put on hold for years.
So please go ahead and branch from master.

HansOlsson, 2017-02-17:
One on hand I believe more visualization will be good, but on the other hand we have an issue that animation objects increase the size of all simulation models (at least for MultiBody systems) - and also makes it more difficult to see the equations for the physical parts.
Thus I believe we should consider how to ensure that we don't significantly impact models if users don't want to visualize.

christiankral, 2017-03-09:
I created the new branch [christiankral_Visualization#2172](https://github.com/modelica/Modelica/tree/christiankral_Visualization%232172) with a test implementation. I tried to avoid all unnecessary variables so that a non-visualized simulation has only little overhead. This was performed by creating conditional classes which are only enabled when a visualization is enabled. These conditional objects contain all the required calculations. So in case of activating the visualizations, an additional phasor object appears. 

I modified the models

- Modelica.Electrical.QuasiStationary.SinglePhase.Examples.SeriesResonance
- Modelica.Electrical.QuasiStationary.SinglePhase.Examples.ParallelResonance

which have now selected visualizations enabled. Please test the implementation and the examples and provide feedback.

In case we agree to implement such kind of animation, it could also be applied to:

- Modelica.Electrical.QuasiStationary.MultiPhase
- Modelica.Electrical.QuasiStationary.Machines 
- Modelica.Magnetic.QuasiStatic.FundamentalWave

christiankral, 2017-03-10:
I also added a voltage and current arrow in the icon layer, in case visualization is enabled. In my implementation the voltage arrow of the icon is drawn from the positive to negative pin. This is the usual way of indicating a voltage arrow in the German speaking parts, see [Zählpfeil](https://de.wikipedia.org/wiki/Z%C3%A4hlpfeil). In the US, the voltag arrow is drawn just the oppositve way from the negative pin to the positive pin, see [Passive Sign Convention](https://en.wikipedia.org/wiki/Passive_sign_convention). 

The actual phasor of the visualization is drawn from the negative to the positive potential. From this point of view, the visualization is more related with the US way of drawing a voltage arrow. However, an alternative, more general implementation were to a + and a - sign, at the positive and negative pin, repspectively. I suppose, this convention is well understood worldwide. Therefore I will switch from the actual voltage arrow implementation in e3b01ba to a + and - sign in the icon.

dietmarw, 2017-03-10:
One thing I noticed and that is bad wrt modularisation is that the visualisation is now dependent on 
`Modelica.Mechanics.MultiBody.*` components. So question is should we rather wait with this feature for a major were we can also move the generic visualisation parts out of `Modelica.Mechanics` and into something like `Modelica.Visulization`?

dietmarw, 2017-03-10:
Alternatively all *new* visualisation items should perhaps already placed under `Modelica.Visualization` under which then existing items can be moved later on. This means the new models would not need a conversion later on.

tbeu, 2017-03-10:
We could copy Modelica.Mechanics.MultiBody.Visualizers (along with their dependencies to Types and Interfaces) to Modelica.Visualizers, mark them as obsolete and use the _new_ visualization items already now.

sjoelund, 2017-03-10:
You could even move them already and create obsolete models extending from the moved models (I guess).

HansOlsson, 2017-03-10:
I don't think we should copy and create obsolete models. 
We can keep existing models in a slightly awkward place, or completely move them (with conversion scripts).

One question is whether all visualizers should be in Modelica.Visualizers - or we view Modelica.Visualizers as base-functionality and MultiBody (and possibly QuasiStationary) has domain-specific visualizers.

dietmarw, 2017-03-10:
Nothing against domain-specific visualisers as long as they only extend from/use something out of `Modelica.Visualization` I'd say.

tbeu, 2017-03-10:
Yes, _major_ means with conversion scripts which is not yet scheduled at all. For that reason a backward-compatible copy/obsolete approach was proposed which could be handled in _minor_, which also is not scheduled, but somehow expected to happen before _major_.

HansOlsson, 2017-03-10:
> Yes, major means with conversion scripts which is not yet scheduled at all. For that reason a backward-compatible copy/obsolete approach was proposed which could be handled in minor, which also is not scheduled, but somehow expected to happen before major.

I understand that - but to me that duplication would add a maintenance cost and be confusing for users, and I don't want to create this as a precedence - e.g. for duplicating QuasiStationary with a better name. I believe it is better to live with the awkwardness of the bad structuring until that point, except if the class is very short - so I don't have a problem with Wavelength spelling correction in Modelica.SIunits.

christiankral, 2017-03-11:
I have absolutely no problem to keep the visualization in an awkward space for now. Many things are not there where we want them to be. That's why were planning to use conversion scripts in the next major release. So for the next major release it will not make much difference whether we move a couple more Modelica classes or not. 

I would therefore like to focus on the advantages rather than on the disadvantages of not having a perfect structure. The visualization of phasors is an important feature to be used in education, where it helps a lot in understanding complex phasors. This is the reason I am coming up with the visualization feature of phasors.

christiankral, 2017-07-04:
After reviewing the comments on this ticket and on ticket #2229 I come to the following conclusions: The visualization of phasor diagrams is actually a 2D issue that shall be supported by the following features:
- Axis labels
- Labels of phasors 
- No 3D view onto a plane representation
- Optional grid lines

None of these features is provided by the ``Modelica.Mechanics.MultiBody.Visualizers`` at the time. Even though I see the potential of the visualization of phasors, I guess we are currently not having the right (open source) tools to provide such a visualization. I thus will move the milestone to NEVER and close this ticket as well as ticket #2229.

closed 2017-07-04 by christiankral