modelica/ModelicaStandardLibrary #2175: Blocks.Tables.CombiTable1Ds seems inconsistent

ceraolo, 2017-02-17:
CombiTable1Ds is a bit confusing.
What is the need of "columns" input parameter? I understand its need in CombiTable1D, not in CombiTable1Ds.
Indeed, since the model has a single input, the first column should indicate the input values and the others the interpolated output. So the number of table columns minus one will determine the number of outputs, and **seems to be no need of the "columns" parameter.**

Moreover, the "table" parameter has as a default:
fill(0.0,0,2) 
and as description
"Table matrix (grid = first column; e.g., table[0,2])"

What is the meaning of having "0" as number of rows in the fill() function? 
**I propose to omit any default and use a more clear description**, e.g.:
"Table matrix (grid = first column; e.g., table[0,1; 1,2; 0,3])"

As an alternative, a default for table could be  [0, 0; 1, 1]
for which output replicates the input.

Moreover, the documentation of the model seems to be pasted from the CombiTable1D's without some necessary modifications. But it is to o early to suggest fixes on it.

beutlich, 2017-02-20:
I see three issues here.

1. Parameter columns: Via parameter columns it can be defined how many columns of the table are interpolated, i.e., you can select less than the available columns or reorder the columns. This is the expected behavior.

1. Default table parameter: In #1137 I also stumbled across the unusual default parameter fill(0.0,0,2). It was set to some reasonable default value by 414466def2e5ead4621e14276d5247658fe67340 but reverted again by c0f62028af0b89d3760cd0af6b9dece7c6acedeb as it raised some more issues.

1. Copy/pasted documentation: I fixed the one error I could find in master and maint/3.2.2. Please let me know if you observed some more errors.

ceraolo, 2017-02-21:
Ah, I did not read documentation correctly. My fault. 
Indeed everything was already clear, and your correction made it clearer.
The "strange" default value is justified by #1137 as well.

But I have still an issue.
You left as a description of table "Table matrix (grid = first column; e.g., table[0,2])". I really don't understand this. 
I made a different proposal, which a like much more. As an alternative, if you prefer, we could put in the description a sub-table of the example used in the documentation, e.g.: 
"Table matrix (grid = first column; e.g., table[0,0; 1,1; 2,4])"
 I proposed differently, just to show that the output does not have to be monotonic.

beutlich, 2017-02-22:
OK, I additionally adapted the comment of the table parameter to match the one of Modelica.Blocks.Sources.TimeTable in  in master and maint/3.2.2.

closed 2017-02-22 by beutlich