modelica/ModelicaStandardLibrary #2181: Core Loss models are missing m modification in coreParameters declaration

JoeRiel, 2017-02-17:
This changed from 3.2.1.  The models Modelica.Electrical.Machines.Losses.DCMachines.Core and ...InductionMachines.Core should use 
`parameter Machines.Losses.CoreParameters coreParameters(final m=1);`
and 
`parameter Machines.Losses.CoreParameters coreParameters(final m=m)`
respectively, in the declaration of coreParameters.

Actually, InductionMachines.Core seems to have a related error. I think it should be
`       parameter Machines.Losses.CoreParameters coreParameters(final m=m);`
`       final parameter Integer m=3 "Number of phases";`
As it is, the parameter m is declared as
`final parameter Integer m=coreParameters.m;`
but no modification is passed to coreParameters, so the value is undecided.

AHaumer, 2017-03-04:
As far as  I see, the parameter m is taken from parameter coreParameters.m but not used. 
coreParameters are propagated in Interfaces.PartialBasicInductionMachine and Interfaces.PartialBasicDCMachine from the machine's parameter records, and coreParameters.m is set final 
parameter Machines.Losses.CoreParameters statorCoreParameters(final m=3, ...
parameter Machines.Losses.CoreParameters coreParameters(final m=1, ...
so I can't figure out the problem?

JoeRiel, 2017-05-08:
The problem is that the m parameter is used in CoreParameters, in the GcRef parameter. It needs a value.

christiankral, 2017-05-10:
>model Core "Model of core losses"
>parameter Machines.Losses.CoreParameters coreParameters;
>final parameter Integer m=coreParameters.m "Number of phases";
>...
>I see nothing that assigns the value of m. It's not assigned in CoreParameters.

In ``Modelica.Electrical.Machines.Losses.InductionMachines.Core`` the parameter ``Machines.Losses.CoreParameters coreParameters`` has no default value assigned. That is correct, as a default value is not required. 

In ``Modelica.Electrical.Machines.Interfaces.PartialBasicInductionMachine`` there is a parameter defined 
```modelica
  parameter Machines.Losses.CoreParameters statorCoreParameters(
    final m=3,
    VRef(start=100),
    wRef=2*pi*fsNominal);
```
which is then propagated to the instance of the core loss model
```modelica
  Machines.Losses.InductionMachines.Core statorCore(
    final coreParameters=statorCoreParameters,
    final useHeatPort=true,
    final turnsRatio=1);
```

The same method is applied in ``Modelica.Electrical.Machines.Interfaces.PartialBasicDCMachine`` where
```modelica
  parameter Machines.Losses.CoreParameters coreParameters(
    final m=1,
    VRef=ViNominal,
    wRef=wNominal);
```
is declared and propagated to the instance of the core loss model
```Modelica
  Machines.Losses.DCMachines.Core core(final coreParameters=coreParameters,
      final useHeatPort=true)
```
So in each case, DC and induction machine, there is obviously a parameter propagated to the instance ``core`` with a strictly assigned parameter value. 

I cannot see an open issue here.

svorkoetter, 2017-05-12:
The usual approach in this case where no particular default is necessarily right is to chose one that is at least plausible, set it as the parameter's _start_ value instead of its actual value (retaining the default _fixed=false_). In that case, a tool is supposed to give a warning that no value was supplied, but that it is using the default. This is what is done, for example, for the R parameter of Modelica.Electrical.Analog.Basic.Resistor, which has a default of 1.

The current situation, whether you think it is right or not, is a regression from 3.2.1.

christiankral, 2017-12-31:
I compared the actual code of Modelica master with tag v3.2.1+build.2. Somewhere on the way from v3.2.1 to master there was a change introduced which is NOT backwards compatible. I have no idea why this change was performed. I cannot see any obvious benefit from the new implementation. Therefore, I am rolling back to the implementation of v3.2.1.

@AHaumer is this change OK with you?  

@JoeRiel is there anything left from your point of view or can we close this ticket?

christiankral, 2017-12-31:
I checked the packages Machines, Magnetic.FundamentalWaves and QuasiStatic.FundamentalWaves and they check all OK with the "old" implementation.

beutlich, 2018-01-02:
> Somewhere on the way from v3.2.1 to master there was a change introduced which is NOT backwards compatible.

Git blame tells it is f7b058c1d03887ed5ccd4a6f9198c154213b86ce for #1783 (@FrenkelJ).

Hence, both the previous fix f7b058c1d03887ed5ccd4a6f9198c154213b86ce as well as the recent commit 6fbbb3c384828eb68ff77c539cfab751b83d6427 have issues.

Generally, it is better to provide such changes as pull request and let us discuss the issues there instead of introducing incomplete commits with subsequent discussions.

christiankral, 2018-01-02:
@beutlich Can you explain what issue and incompleteness, respectively, the recently introduced commit causes?

beutlich, 2018-01-02:
Apparently f7b058c1d03887ed5ccd4a6f9198c154213b86ce tried to fix #1783 but introduced this issue as reported by @JoeRiel. By reverting f7b058c1d03887ed5ccd4a6f9198c154213b86ce we still face the original issue as reported by @FrenkelJ.

AHaumer, 2018-01-03:
Should be applied to maintenance 3.2.2, too

commit 191beb06 (2018-01-03): resolves #2181 (discussed with Christian)
commit ad5a9459 (2018-01-03): resolves #2181
closed 2018-01-03 by AHaumer