modelica/ModelicaStandardLibrary #2184: StrayLoad loss model in MSL.Electrical.Machines, speeds that cross zeros

markusandres3DS, 2017-02-21:
### Model: 
Modelica.Electrical.Machines.Losses.InductionMachines.StrayLoad

### Problem descritpion: 
Picking a parameter "power_w" that is not a full number (in my case I chose 0.4) gives problems when the speed is going through zero during simulation. 

### Suggested fix:
A possible fix seems to be the addition of a noEvent additionally to the smooth operator. Could be e.g. the following:

if (strayLoadParameters.PRef <= 0) then
    tau = 0;
  else
    tau = -strayLoadParameters.tauRef*(iRMS/strayLoadParameters.IRef)^2*
      smooth(1, noEvent(if w >= 0) then +(+w/strayLoadParameters.wRef)^strayLoadParameters.power_w 
                                                        else -(-w/strayLoadParameters.wRef)^strayLoadParameters.power_w);
  end if;

AHaumer, 2017-03-04:
To be honest, after looking into that I'm unsure about the smooth ...
I think we should replace the noEvent by abs (which triggers no event!), i.e.
tau = -strayLoadParameters.tauRef*(iRMS/strayLoadParameters.IRef)^2*
           (abs(w)/strayLoadParameters.wRef)^strayLoadParameters.power_w

AHaumer, 2017-03-04:
The stray load loss model and the permanent magnet loss model should have the same issue:
Modelica.Electrical.Machines.Losses.InductionMachines.StrayLoad
Modelica.Electrical.Machines.Losses.InductionMachines.PermanentMagnetLosses
Modelica.Electrical.Machines.Losses.DCMachines.StrayLoad
Modelica.Magnetic.QuasiStatic.FundamentalWave.Losses.StrayLoad
Modelica.Magnetic.QuasiStatic.FundamentalWave.Losses.PermanentMagnetLosses
We should add an example for testing to ModelicaTest and then test the new implementation comparing it with respect to the old implementation.

AHaumer, 2017-03-05:
resolved by SHA-1: f8c9490b31f18259cb269d563eb5c12e5b7935ae for maint/3.2.2
resolved by SHA-1: 2d0c14160607321c13c682d21c83b724b29a99d1 for master

closed 2017-03-05 by AHaumer

markusandres3DS, 2017-03-06:
As abs() results in the same as a noEvent() for that case, it seems to be an equally good solution to me.

commit f8c9490b (2017-03-05): resolves #2184 for maint/3.2.2