modelica/ModelicaStandardLibrary #2186: Questionable constantVoltage Symbol

ceraolo, 2017-02-22:
In MSL Electrical.Analog.Sources symbols are generally directly drawn from International standards, with slight modifications.  The applicable standards are IEC 60375 and IEC 60617

For current sources, clause 8.2.1 of IEC 60375 defines two symbols (a and b); of them, only version (a) is standardised in IEC 60617.
The current source symbol in MSL is a slight (and well acceptable) modification of version (a). 

For voltage sources, clause 8.1.1 of IEC 60375 defines two symbols (a and b); of them, only version (a) is standardised in IEC 60617.
In MSL, signal voltage source, is a slight (and well acceptable) modification of version (a).

**However the "ConstantVoltage" Symbol is totally different: it is the one commonly used for electro-chemical cells, not ideal voltages.**

**_I propose to change the ConstantVoltage Symbol, and to use one identical to the present "SignalVoltage" symbol, except for the obvious absence of the input arrow._**

dietmarw, 2017-02-22:
Explanation for the milestone setting. Though not a backward compatible change I would say it has quite an influence on the user experience ans should probably not be changed with a minor release.

ceraolo, 2017-02-22:
I don't want to insist too much on a cosmetic change.
But MSL users are already accustomed with the symbol, since it is the one already used for the signalVoltage.
In my opinion is a good point of Modelica that it uses symbols already very near to the standard ones (I would even say as near as possible) with few exceptions.
Removing this exception, even in the next minor revision, would just help to make diagrams more professional-looking. 
I hope some other users of electric diagrams give their opinion here.

beutlich, 2017-02-27:
This also applies to Modelica.Electrical.Spice3.Sources.V_constant.

beutlich, 2017-02-27:
It already is done right in Modelica.Electrical.MultiPhase.Sources.ConstantVoltage.

kristinmajetta, 2017-03-01:
For me it would be ok to redraw the symbol even in a minor version. I do not understand why the new symbol would influence on the user experience.

dietmarw, 2017-03-01:
Well the user's models suddenly would look all different. It is the visual user experience I was talking about. But you are the lib officer, so it is your call @kristinmajetta :smirk:

ceraolo, 2017-03-01:
Well, I asked people to give their opinions about this ticket because I thought that they could be good samples of the average final users' opinion about the change.

But in case, as it seems, what impact the change on the user experience is already, _a priori_, known  to the lib officers, I'm totally wrong, and I must apologise with @kristinmajetta to have invited him to a useless activity.

ceraolo, 2017-11-17:
So is it decided to accept my suggestion?
In case, may I ask to update the symbol in the trunk version? From OpenModelica libraries it appears that this change has not been implemented yet.

beutlich, 2017-11-17:
Sometimes an image says more than words. Here is how the ConstantVoltage icon currently looks in Multiphase, Analog and Spice3.
[image]
To sum it up: @ceraolo says it should be changed, @dietmarw says it is not backward compatible from the usage point of view and @kristinmajetta simply does not know which way to go. Thus, it still is not decided yet.

kristinmajetta, 2018-04-23:
changed symbol of Analog.Souces.ConstantVoltage according to suggestion from ceraolo
SHA-1: 79081ba8f80b6ff08add4ab6941cd6c99d3cb8a0

beutlich, 2018-04-23:
Edit milestone since 79081ba8f80b6ff08add4ab6941cd6c99d3cb8a0 went to master.

beutlich, 2018-04-23:
Should Modelica.Electrical.Spice3.Sources.V_constant also be adapted?

kristinmajetta, 2018-04-23:
Yes it should. But since it was not a task in that ticket I thought it would be better to open a new ticket for that issue with Modelica.Electrical.Spice3.Sources.V_constant.

max-privato, 2018-04-23:
Thanks a lot everybody.
Several people (among which myself) use Modelica in teaching, and having it standard compliant in terms of graphical symbols is an important advantage. This ticket helps in this direction.
I will sure download Modelica master very soon.

@kristinmajetta, do you think it is me that should open the new ticket on Modelica.Electrical.Spice3.Sources.V_constant, or it doesn't matter?

tobolar, 2018-04-23:
If its fine to change icons in Modelica.Electrical.Spice3 why it should be postponed for the next major in #2486?

closed 2018-04-24 by beutlich