modelica/ModelicaStandardLibrary #2188: PartialBooleanBlock rectangle line too thick?

ceraolo, 2017-02-26:
The model 
  Modelica.Blocks.Icons.PartialbooleanBlock
contains a rectangle having: 
          lineThickness=5.0,
Is this value true? It seems too large ( I intend 5 logical millimetres, i.e. 5 mm on a PC with default 96 logical dpi and a screen having actually 96 DPI).

Dymola does not draw this rectangle at all (maybe  with a logic stating not to draw rectangles when 
borderPattern=BorderPattern.Raised) OM, which starting from two days ago is DPI-aware also in drawing lines, draws the rectangle using a huge line.

What is the intended behaviour? Maybe the rectangle outline may just be omitted (as in Dymola's behaviour)?

beutlich, 2017-02-27:
I believe this is a specification issue if the rectangle line is to be drawn if the border pattern is note None. However, the MLS 3.3rev1 already states in 18.6.1.2
> The border pattern attributes  Raised, Sunken and Engraved represent
frames which are rendered in a tool-dependent way.

Thus it is not obvious if we should clarify the tool-dependent way. @HansOlsson Do you think this needs to be clarified more? If yes, we would need a Trac ticket then.

ceraolo, 2017-02-27:
Even in the case in which no change in the MLS is needed (because of the tool-dependent nature of border pattern) there is a MSL issue, since 5 is a VERY large thickness, as I said.
If tool dependency means that a tool could can decide whether draw the rectangle or not, then the thickness should be wiser, e.g. 0.25 or  0.5  (mm).

ceraolo, 2017-03-25:
While waiting for possibile specification modifications, is it possibile.to change the thickness in the MSL trunk? It is needed anyway.

beutlich, 2017-03-27:
Resolved by 4a80406f459d76d241bce4f5640d7fe3a9dfadc1 in master branch.

Sorry for the delay. I was hoping to receive some feedback from the library officer first, but he kept silent.

ceraolo, 2017-03-28:
There are other model icons with 5mm thickness.
This is my list:

Blocks.Icons.PartialDiscreteBlock (fixed)
Blocks.Icons.DiscreteBlock

Blocks.Interfaces.PartialBooleanMISO
Blocks.Interfaces.PartialBooleanThresoldComparison
Blocks.Interfaces.PartialBooleanComparison
Blocks.Interfaces.PartialBooleanSISO_small
Blocks.Interfaces.PartialBooleanMISO
Blocks.Interfaces.PartialIntegerSISO
Blocks.Interfaces.PartialIntegerMISO
Blocks.Interfaces.DiscreteBlockIcon
Blocks.Interfaces.PartialBooleanBlockIcon

Blocks.Sources.RealExpression
Blocks.Sources.IntegerExpression
Blocks.Sources.BooleanExpression

beutlich, 2017-03-28:
Blocks.Icons.DiscreteBlock should also be fixed by 4a80406f459d76d241bce4f5640d7fe3a9dfadc1.

beutlich, 2017-03-28:
Resolved 4687753a886fb29b489f36f9587536f38abdb01d in master branch.

closed 2017-03-28 by beutlich