modelica/ModelicaStandardLibrary #2190: Add periodic continuation for BooleanTable and IntegerTable

christiankral, 2017-02-27:
Periodic continuation would also make sense for
- ``Modelica.Blocks.Sources.BooleanTable``
- ``Modelica.Blocks.Sources.IntegerTable``

The feature of periodic continuation is currently only implemented for real value tables based on external objects, see #2120.

beutlich, 2017-02-27:
What do you think of this [IntegerTable](https://gist.github.com/tbeu/6fe254d4bc84e8abb74a110775954ce0) as replacement? Still, the documentation is not correct, as there is no longer an assertion in case of non-Integer table values.

christiankral, 2017-02-27:
My original issue was having a periodic boolean table (implemented in MSL). I added ``IntegerTable`` just because it is also there in MSL it makes sense to have identical implementations.

beutlich, 2017-02-27:
The BooleanTable is different from IntegerTable or TimeTable since it accepts a vector as table only and alternates the output whenever a sample point is met. You would need to define what the expected behavior of a periodic boolean table is where the number of sample points is odd!

beutlich, 2017-03-13:
@christiankral What shall be the expected behavior of periodic BooleanTable with odd number of points?

christiankral, 2017-03-14:
The BooleanTable uses a parameter ``table`` which toggles the output. I interpret the period as 
```modelica
T = table[size(table,1)]-table[1];
```
So to my understanding the periodic behavior is well determined for odd number of table points. Please consider the following examples: 

[image]

Odd number of table points ``table = {0,0.4,1}``

Even number of table points ``table = [0,0.4,0.6,1]``
A) The table output is continued asymmetrically, i.e, after one "period" the output is inverse to the previous "period", which means that the actual period is actually 2*T, not T
B) The table output is continued periodically; there is not switching event at the last table point

I propose to implement variant B) as this matches the term period closer than case A)

beutlich, 2017-04-05:
The solution based on CombiTimeTable is blocked by #1771. :boom:

beutlich, 2017-04-05:
@christiankral What do you think of the general idea of utilizing CombiTimeTable for the periodic extrapolation feature?

christiankral, 2017-04-06:
From a practical point of view this is a good idea and I am in favor of it. Additionally, it is a modular concept which very much follows the object oriented approach of Modelica.

beutlich, 2017-04-07:
Resolved in master by b2b043544721b82b1b6098e609bfc3cefac3e096.

closed 2017-04-07 by beutlich

christiankral, 2017-04-09:
Thanks a lot for your implementation.

beutlich, 2017-04-09:
Attached are the updated Win libraries in case you want to test the new feature.

[T2190.zip](https://github.com/modelica/Modelica/files/908411/T2190.zip)