modelica/ModelicaStandardLibrary #2191: Add blocks for inverese of log2 and log10

christiankral, 2017-02-27:
In ``Modelica.Blocks.Math`` there are blocks of all trigonometric functions including their inverse functions, e.g. ``Cos`` and ``Acos``. In the same manner ``Log`` and ``Exp`` are implemented. Unfortunately, for ``Log2``and  ``Log10`` there do not exist inverse blocks. These inverse blocks shall also be implemented as 

- y = 2^u 
- y = 10^u

 Otherwise, some graphical implementations are a bit clumsy. 

Possible names of the missing blocks are
- ``Power2`` 
- ``Power10``

beutlich, 2017-02-27:
Where do you see Log2?

christiankral, 2017-02-27:
You're right. Log2 shall be added as well.

sjoelund, 2017-02-27:
Would it be sufficient with a general `Power` block instead of a `Power2`, `Power10` block?

christiankral, 2017-02-27:
Yes, this also serves the purpose and is more flexible. I am favor of implementing ``Power``.

HansOlsson, 2017-02-28:
I don't think we should add Log2 - but I am in favor of generalizing Log and Exp (or adding Power).

One possibility would be to add a parameter base (default `e` to Log or `10` to Log10); another option would be to add a new block.
Similarly we should have base for Power/Exp-block. There are some minor numerical issues (i.e. whether log(e)=1 or if that will add an offset).
Having the base as an input might be possible, but seems excessive.

christiankral, 2017-02-28:
Adding the base is more flexible approach. I am in favor of providing a generalized ``Power`` and ``Log`` function. 

I would also not go for a base input, but for a base parameter (for example, with default ``Modelica.Constants.e``).

beutlich, 2017-03-27:
@christiankral Are you going to provide a pull request for this issue? Otherwise I will assign it to the library officer.

christiankral, 2017-03-27:
OK I will provide a pull request.

beutlich, 2017-04-04:
Resolved by d23413c2625bfc44c86a9a5a27192a1bb79c6767.

closed 2017-04-04 by beutlich