modelica/ModelicaStandardLibrary #2193: Problems with equations from IdealGTOThyristor resulting in infinite loop

agnesramle, 2017-02-28:
In the models
* Modelica.Electrical.PowerConverters.Examples.DCAC.MultiPhaseTwoLevel.MultiPhaseTwoLevel_RL
* Modelica.Electrical.PowerConverters.Examples.DCAC.SinglePhaseTwoLevel.SinglePhaseTwoLevel_RL
* Modelica.Electrical.PowerConverters.Examples.DCDC.HBridge.HBridge_DC_Drive

we have the following mixed linear system
--- Torn mixed linear system (Block 3) of 4 iteration variables and 7 solved variables ---
Coefficient variability: discrete-time
Torn variables:
  inverter.transistor_n.i
  inverter.diode_p.i
  inverter.diode_p.v
  fundamentalWaveVoltage.u
  inverter.diode_n.v
  inverter.diode_n.i
  inverter.transistor_p.i

Iteration variables:
  inverter.transistor_p.s
  inverter.transistor_n.s
  inverter.diode_p.s
  inverter.diode_n.s

Solved discrete variables:
  inverter.transistor_p.off
  inverter.transistor_n.off
  inverter.diode_p.off
  inverter.diode_n.off

Torn equations:
  inverter.transistor_n.i := inverter.transistor_n.s * (if inverter.transistor_n.off then inverter.transistor_n.Goff else 1) + inverter.transistor_n.Goff * inverter.transistor_n.Vknee
  inverter.diode_p.i := inverter.diode_p.s * (if inverter.diode_p.off then inverter.diode_p.Goff else 1) + inverter.diode_p.Goff * inverter.diode_p.Vknee
  inverter.diode_p.v := - inverter.transistor_p.s * (if inverter.transistor_p.off then 1 else inverter.transistor_p.Ron) - inverter.transistor_p.Vknee
  fundamentalWaveVoltage.u := inverter.diode_p.v + constantVoltage_p.p.v
  inverter.diode_n.v := - fundamentalWaveVoltage.u + constantVoltage_n.n.v
  inverter.diode_n.i := inverter.diode_n.s * (if inverter.diode_n.off then inverter.diode_n.Goff else 1) + inverter.diode_n.Goff * inverter.diode_n.Vknee
  inverter.transistor_p.i := inverter.transistor_p.s * (if inverter.transistor_p.off then inverter.transistor_p.Goff else 1) + inverter.transistor_p.Goff * inverter.transistor_p.Vknee

Continuous residual equations:
 ( - inverter.diode_n.v) = inverter.transistor_n.s * (if inverter.transistor_n.off then 1 else inverter.transistor_n.Ron) + inverter.transistor_n.Vknee
    Iteration variables: inverter.transistor_p.s
  inverter.diode_p.v = inverter.diode_p.s * (if inverter.diode_p.off then 1 else inverter.diode_p.Ron) + inverter.diode_p.Vknee
    Iteration variables: inverter.transistor_n.s
  inverter.diode_n.v = inverter.diode_n.s * (if inverter.diode_n.off then 1 else inverter.diode_n.Ron) + inverter.diode_n.Vknee
    Iteration variables: inverter.diode_p.s
  inductor.i + (- inverter.diode_n.i) + inverter.diode_p.i + inverter.transistor_n.i + (- inverter.transistor_p.i) = 0
    Iteration variables: inverter.diode_n.s

Discrete equations:
  inverter.transistor_p.off := inverter.transistor_p.s < 0 or not inverter.andCondition_p.y
  inverter.transistor_n.off := inverter.transistor_n.s < 0 or not inverter.andCondition_n.y
  inverter.diode_p.off := inverter.diode_p.s < 0
  inverter.diode_n.off := inverter.diode_n.s < 0

Jacobian:
  |1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, - (if inverter.transistor_n.off then inverter.transistor_n.Goff else 1), 0.0, 0.0|
  |0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, - (if inverter.diode_p.off then inverter.diode_p.Goff else 1), 0.0|
  |0.0, 0.0, -1.0, 0.0, 0.0, 0.0, 0.0, - (if inverter.transistor_p.off then 1 else inverter.transistor_p.Ron), 0.0, 0.0, 0.0|
  |0.0, 0.0, -1.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0|
  |0.0, 0.0, 0.0, -1.0, -1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0|
  |0.0, 0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, - (if inverter.diode_n.off then inverter.diode_n.Goff else 1)|
  |0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 1.0, - (if inverter.transistor_p.off then inverter.transistor_p.Goff else 1), 0.0, 0.0, 0.0|
  |0.0, 0.0, 0.0, 0.0, -1.0, 0.0, 0.0, 0.0, - (if inverter.transistor_n.off then 1 else inverter.transistor_n.Ron), 0.0, 0.0|
  |0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, - (if inverter.diode_p.off then 1 else inverter.diode_p.Ron), 0.0|
  |0.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 0.0, 0.0, - (if inverter.diode_n.off then 1 else inverter.diode_n.Ron)|
  |1.0, 1.0, 0.0, 0.0, 0.0, -1.0, -1.0, 0.0, 0.0, 0.0, 0.0|

 The right hand side of this system is independent of the switches and if we have a little bit of noise for the state variable inductor.i (e.g.. is -2.0000006918309519E-003 instead of -2E-003) at the time point where  inverter.andCondition_p.y is false and inverter.andCondition_n.y is true, the iteration variable  inverter.transistor_n.s will be bouncing between being equal to exactly 0 and being slightly negative (in the size of ~-1e-10) resulting in an infinite loop for the switches.

christiankral, 2017-03-02:
I suppose this issue occurs in JModelica. I tested the three models with JModelica trunk version 2017-03-01 and all three models failed. 

I also tested the three models with OpenModelica and the examples work OK, even though I get some warning which may be related with the non-linear solver. The models work with no complaints in Dymola.

I also tested the examples (without inductor)

- Modelica.Electrical.PowerConverters.Examples.DCAC.MultiPhaseTwoLevel.MultiPhaseTwoLevel_R
- Modelica.Electrical.PowerConverters.Examples.DCAC.SinglePhaseTwoLevel.SinglePhaseTwoLevel_R

and they work OK in JModelica, OpenModelica and Dymola.

So before I am trying to come up with some physical modeling fixes, I would like to hear your opinion. When I see all these sets of equations and the Jacobian, I am not quite sure in what direction I shall investigate the issue. The jittering of the state variable s somehow suggests to me that the interaction of the ``Modelica.Electrical.Analog.Ideal.IdealDiode`` with ``Modelica.Electrical.Analog.Ideal.IdealGTOThyristor`` causes the problem? Do you agree and/or do you have a different suggestion?

agnesramle, 2017-03-03:
Yes, I do agree. When this problem occurs in JModelica the rhs of the problem is
<vector name="b">      -1.0000000000000000E+002,        0.0000000000000000E+000,        1.0000000000000000E+002,       -2.0000006918309519E-003</vector>
and the rhs is independent of the switches.
The Jacobian in the case when inverter.transistor_n.off is false the Jacobian looks like:
 <matrix name="A">
                      -1.0000000000000000E+000,       -1.0000000000000001E-005,        0.0000000000000000E+000,        0.0000000000000000E+000;
                      -1.0000000000000000E+000,        0.0000000000000000E+000,       -1.0000000000000000E+000,        0.0000000000000000E+000;
                       1.0000000000000000E+000,        0.0000000000000000E+000,        0.0000000000000000E+000,       -1.0000000000000001E-005;
                      -1.0000000000000001E-005,        1.0000000000000000E+000,        1.0000000000000001E-005,       -1.0000000000000000E+000;
              </matrix>
and the Jacobian when inverter.transistor_n.off is true looks like:
<matrix name="A">
                      -1.0000000000000000E+000,       -1.0000000000000000E+000,        0.0000000000000000E+000,        0.0000000000000000E+000;
                      -1.0000000000000000E+000,        0.0000000000000000E+000,       -1.0000000000000000E+000,        0.0000000000000000E+000;
                       1.0000000000000000E+000,        0.0000000000000000E+000,        0.0000000000000000E+000,       -1.0000000000000001E-005;
                      -1.0000000000000001E-005,        1.0000000000000001E-005,        1.0000000000000001E-005,       -1.0000000000000000E+000;
              </matrix>.
This yields in the false case the following iteration variable values:
<vector name="ivs">       1.0000000000000000E+002,       -3.4591547585523081E-010,       -1.0000000000000000E+002,        3.4591547594289505E-010</vector>
and in the true case:
<vector name="ivs">       1.0000000000000000E+002,        0.0000000000000000E+000,       -1.0000000000000000E+002,        6.9183095134170243E-010</vector>

The variable values for neither the iteration variables nor the torn variables vary very much even though the switches switch.

christiankral, 2017-03-04:
I created a new repository https://github.com/christiankral/PowerConvertersTesting to perform some tests. So for now, I made a little step ahead. Please consider the branch ``InductorGTO-Diode``:

I added inductors between each GTO and diode. This is applied to the single and multi phase inverter. The default values of the inductance are set to L = 1E-5H. 

[image]

[image]

When I test the models

- PowerConvertersTesting.Examples.DCAC.SinglePhaseTwoLevel.SinglePhaseTwoLevel_RL
- PowerConvertersTesting.Examples.DCAC.MultiPhaseTwoLevel.MultiPhaseTwoLevel_RL
- PowerConvertersTesting.Examples.DCDC.HBridge.HBridge_DC_Drive

they simulate in JModelica.

Physically, there do always exist some stray (leakage) inductances between (discrete) components. So far, I tried to avoid any of those, in the basic PowerConverters library, provided in the MSL, because they do also cause several drawbacks:
- the inductances are usually not known if you haven't designed the inverter
- the additional inductors may introduce some additional small time constants which most likely results in a stiff system of differential equations
- for consistency reasons, stray inductors would have to be considered on many additional locations then which makes the models even more difficult to parameterize and handle
- additional effects are introduced which are not "backwards compatible" (see reverse DC currents in next figure)

[image]

christiankral, 2017-03-04:
Since the in an inverter model the GTO and diode are always used in anti parallel connection we could think about an additional implementation: a transistor model with integrated anti parallel diode

The original configuration of the GTO and diode together have four different states 

no. | condition | GTO | diode   
---  | ---: | --- | --- 
1 | GTO not turned on, v_GTO > 0 | off | off 
2 | GTO turned on | on | off 
3 | reverse conduction | off | on 
4 | theoretical only | on | on 

So obviously, the fourth state could be omitted in an integrated model. This might then help to avoid jittering effects.

AHaumer, 2017-03-05:
I added a sub-package PowerModule to PowerConvertersTesting with a PowerModule (made of components: 2 switches + 2 free-wheekng diodes) and a Test example, then I started to implement a PowerModule with equations and the above mentioned states (but not yet ready).

AHaumer, 2017-03-05:
This is the PowerModule built from components:
[image]
It seems that we have only a reduced number of states:
[image]
I'll double-check that whether it's true or not, and it seems that this could ease the tool's job by reducing the number of states from 16 combinations of 4 booleans to 5 meaningful states (see table above).

AHaumer, 2018-05-18:
I think this issue needs further discussion (maybe a tool issue?)

christiankral, 2020-02-15:
@agnesramle could you please possibly give us an update on this issue to let us know if you could already resolve this issue or not.

modelonrobinandersson, 2020-02-17:
Hi @christiankral, this issue still exists with MSL 3.2.3 (build 3).