modelica/ModelicaStandardLibrary #2198: Add calculation of AC and DC power to connector interfaces in PowerConverters

christiankral, 2017-03-05:
Currents and voltages are already defined in the interfaces, but it also makes sense to have power in PowerConverters available. This way it can be avoided to use additional power sensors in experiments in order to just visualize power in a result plot.

AHaumer, 2017-03-05:
You mean adding power calculation to components, not to connectors, I assume. 
This makes sense. We could also consider calculating mean power over a switching cycle.

christiankral, 2017-03-05:
The power calculation refers to the interfaces of

- Modelica.Electrical.PowerConverters.Interfaces.ACDC
- Modelica.Electrical.PowerConverters.Interfaces.DCAC
- Modelica.Electrical.PowerConverters.Interfaces.DCDC

The power terms to be calculated shall be instantaneous power.

Example:

```modelica
  partial model DCtwoPin1 "Positive and negative pins of side 1"
    Modelica.Electrical.Analog.Interfaces.PositivePin dc_p1
      "Positive DC input";
    Modelica.Electrical.Analog.Interfaces.NegativePin dc_n1
      "Negative DC input";
    Modelica.SIunits.Voltage vDC1=dc_p1.v - dc_n1.v "DC voltage side 1";
    Modelica.SIunits.Current iDC1=dc_p1.i "DC current side 1";
  end DCtwoPin1;
```

Add
```modelica
  Modelica.SIunits.Power powerDC1 = vDC1*iDC1 "DC power side 1";
```

christiankral, 2017-03-05:
Fixed.

closed 2017-03-05 by christiankral