modelica/ModelicaStandardLibrary #2203: Fix documentation of Modelica.Blocks

HansOlsson, 2017-03-06:
When writing a reply to #2168  I noticed that some blocks have a scalar input but are described as if they have an input with many elements. That is confusing and likely a left-over from Modelica standard library version 1 (when they had an actual vector as input).

At least Modelica.Blocks.Math.Sqrt Modelica.Blocks.Math.Asin Modelica.Blocks.Math.Acos have this problem, but it is likely that there are more.

HansOlsson, 2017-03-06:
Modelica.Blocks.Continuous.Integrator and Modelica.Blocks.Continuous.LimIntegrator are also described as element-wise operations; but have a single scalar input.

beutlich, 2017-04-03:
@MartinOtter When will you be able to deal with it?

PR #2265 (2017-06-15): Improve documentation of Blocks
Resolve #2203  by describing scalars as scalars and not as vectors. 
Resolve #2168 by not limiting complex Acos and Asin, and by only restricting Log to non-zero values.
closed 2017-06-15 by MartinOtter