modelica/ModelicaStandardLibrary #2207: Changing constants (2018 edition)!

HansOlsson, 2017-03-08:
This seems similar to #1813 - but is different since it is the **final** change for many of the constants.
This means that e.g. elementary charge and Avogrado's constant (similarly as the speed of light) will be defined to have specific values - instead of being measured.

The branch HansOlsson:HansOlsson-codata2017 contain the best available data so far based on http://www.bipm.org/utils/common/pdf/si-brochure-draft-2016b.pdf

Road-map for the change is: http://www.bipm.org/utils/common/pdf/SI-roadmap.pdf#page=3
The values are not yet accepted, so the change cannot be merged yet.

thorade, 2017-03-08:
Also related: #1241 and #1266

beutlich, 2019-05-21:
The 9th edition of the SI brochure is available at https://www.bipm.org/utils/common/pdf/si-brochure/SI-Brochure-9-EN.pdf

closed 2019-05-28 by beutlich

thorade, 2021-11-08:
Related: https://github.com/usnistgov/REFPROP-issues/issues/239

PR #2774 (2019-05-28): Update of values for Modelica.Constants according to codata2017
Update of values for Modelica.Constants
Closes #2207 
The values are already more correct that previous ones, and from 2019-05-20 they will be likely be final; except for G and T_zero (on the other hand the triple point for water (suitably defined) will be 273.16K).

Watch live: https://www.youtube.com/watch?v=qA67T7FPBME
commit 5cb22fea (2019-05-28): Update of values for Modelica.Constants according to codata2017 (#2774)

* Preliminary update for new SI-system

* Update

* Update

* RenamedToQ

* Update Constants.mo

Based on: https://www.bipm.org/utils/en/pdf/CGPM/Draft-Resolution-A-EN.pdf
(Which changes SI-system on 2019-05-20, and will be discussed today 2018-11-16.)

Note that G is still determined experimentally, and I don't know if that was changed in codata 2017 (doesn't seem that way; since G is really tricky to measure and unrelated to the other constants.)
Closes #2207

* Update Constants.mo

Remove previous values.

* References [... cut]