modelica/ModelicaStandardLibrary #2211: Tool-neutral Modelica.Utilities.System.exit

HansOlsson, 2017-03-14:
The function `Modelica.Utilities.System.exit` needs to have a tool-specific implementation (instead of directly calling `exit` a tool normally wants to intercept this and do an orderly shut-down e.g. to allow users to save files).
However, Modelica Standard Library, including the C-library ModelicaExternal is intended to be tool-neutral.

The solution is move the implementation of exit to a function in ModelicaServices (that vendors will need to implement - the default implementation can call `exit`) and have `Modelica.Utilities.System.exit` call the function in ModelicaServices.

Migrated from: https://trac.modelica.org/Modelica/ticket/1952

beutlich, 2017-03-14:
Currently there is no vendor-specific C library for ModelicaServices. We could move it to C library ModelicaUtilities instead and treat it similar as ModelicaAllocateString etc.

HansOlsson, 2017-03-23:
> Currently there is no vendor-specific C library for ModelicaServices. We could move it to C library ModelicaUtilities instead and treat it similar as ModelicaAllocateString etc.

The point was to let each tool implement their own solution; if the tool-specific version involves a C-library or not doesn't really matter - since it is up to the tool vendor.

For the default version of ModelicaServices I agree that having a C library seems a bit excessive - we could add a default implementation that does one of the following:

1. Just prints a message, perhaps using `algorithm assert(false, "Need to implement exit");`
2. Use `external "C" exit(val); `

sjoelund, 2017-03-23:
Well, exit(0) should be more like terminate() than assert, right? External functions can already call ModelicaError if they like. They cannot call anything like ModelicaTerminate.

HansOlsson, 2017-03-23:
> Well, exit(0) should be more like terminate() than assert, right?

True, but a bit different. So, the assert was primarily to indicate that it wasn't implemented - not an actual implementation. Using external "C" exit(val) is an actual implementation - just suboptimal.

beutlich, 2017-04-18:
Resolved in master.

closed 2017-04-18 by beutlich