modelica/ModelicaStandardLibrary #2219: Improve ReleaseNotes

beutlich, 2017-03-24:
@christiankral Recently you edited the release notes of your libraries, i.e.,
* Modelica.Magnetic.FundamentalWave.UsersGuide.ReleaseNotes
* Modelica.Electrical.MultiPhase.UsersGuide.ReleaseNotes
* Modelica.Electrical.Machines.UsersGuide.ReleaseNotes
* Modelica.Electrical.PowerConverters.UsersGuide.ReleaseNotes
* Modelica.Electrical.QuasiStationary.UsersGuide.ReleaseNotes

but touched MSL v3.2.2 this way by adding new notes to it.

Can you please 
* Introduce a new version of MSL (as long as it is what the next version will be), say MSL 3.x.x, see e.g., Modelica.Magnetic.QuasiStatic.FundamentalWave.UsersGuide.ReleaseNotes
* Add clickable links to the corresponding tickets, see e.g., [Modelica.Electrical.QuasiStationary.UsersGuide.ReleaseNotes](http://doc.modelica.org/help/Modelica_Electrical_QuasiStationary_UsersGuide.html#Modelica.Electrical.QuasiStationary.UsersGuide.ReleaseNotes) of MSL v3.2.2
* Use `h4` headings (instead of `h5`), see e.g., Modelica.Electrical.Digital.UsersGuide.ReleaseNotes or Modelica.Fluid.Dissipation.UsersGuide.ReleaseNotes.

Thank you for all your contributions.

christiankral, 2017-03-26:
From I understanding the changes you refer to have been applied in master and maint/3.2.2. This is the reason why I touched the v3.2.2 release notes. If we need to distinguish different build versions of v3.2.2 in the release notes this is OK, but it needs to be communicated.

beutlich, 2017-03-27:
We had different version releases in the past (was not my preferred idea) on MSL v3.2.1.

christiankral, 2017-04-01:
So how shall we proceed here? For the changes applied to both master and maint/3.2.2 we should establish a general rule on how we handle this. 

I think the main question is here: how much information on the change dates is required. If we keep the references to issues on GitHub, we do not have to write down the change date of each change. For the user, the main interest may be which versionBuild does contain which change of the library.

OK, let me make some proposals; my preference here is proposal 2:

### Proposal 1
We always use the latest update date (this is what I did so far); drawback: no structuring of versionBuild numbers
```
Version 3.2.2, 2017-04-01
* Added component C
* Additional bug fix in component B
* Bug fix in component B
```

### Proposal 2
We structure the versions by the versionBuild number and the latest change date, this proposal it is fully compatible with the tags we are using: version numbers + versionBuild
```
Version 3.2.2, build 4, 2017-03-01
* Added component C
* Additional bug fix in component B
Version 3.2.2, build 3, 2016-12-24
* Bug fix in component B
```

### Proposal 3

Add the version date to each change
```
Version 3.2.2, build 4
* 2017-03-01: Added component C
* 2017-01-07: Additional bug fix in component B
Version 3.2.2, build 3
* 2016-12-24: Bug fix in component B
```

Other proposals?

dietmarw, 2017-04-01:
I just like to point out that in an earlier discussion it was already found that the `versionBuild` should be retired in sake of semantic versioning. But since we still don't have a proper release procedure in place and neither some strategy I think it is best to basically contain as much as possible information and not concentrate of the to be retired `versionBuild`. 

Since so far the maintenance release (i.e., 3.2.2+) is basically a rolling release proposal 1 would probably be fine. But how many date stamps would you include? Each time you make a change? Anything else would require a proper release cycle. Again, due to lack of maintenance strategy difficult to solve.  But then you would only have proper release notes in real releases right. And it does not make sense to add the git log to the release notes of non-releases.

christiankral, 2017-04-02:
In proposal 1 so far I include only one time stamp. In the way I did it, the time stamp represents the latest change date.

beutlich, 2017-04-18:
For 43230160ccd80f06cd880a5ec5ba0925372bc432 I temporarily added 3.x.x to the release notes.

ThomasBoedrich, 2017-06-06:
Removed Label "Magnetic.FluxTubes", since library is not affected by this issue

christiankral, 2017-07-04:
From the discussion of this ticket I conclude, that we stick with the current procedure. It, however, makes sense to indicate changes with respect to previous release versions by version number 3.X.X as shown in 4323016. This procedure allows a to make decision later on whether the next release becomes an maintenance or major release. 

I added some comments on this procedure in ``Modelica.UsersGuide.Conventions.UsersGuide.RevisionHistory`` (see a70dee0 and 982885f)

closed 2017-07-04 by christiankral