modelica/ModelicaStandardLibrary #2224: Missing comments in Modelica.Electrical.Analog.Examples.OpAmps.SignalGenerator

beutlich, 2017-03-28:
Parameters R1, R2, R and C of Modelica.Electrical.Analog.Examples.OpAmps.SignalGenerator miss a descriptive comment.

beutlich, 2017-04-18:
These parameter comments are also missing for all other OpAmps examples, except VoltageFollower.

kristinmajetta, 2017-04-21:
descriptive comments added for parameters R1, R2, R and C of Modelica.Electrical.Analog.Examples.OpAmps.SignalGenerator.
Model description improved and standard value of parameter Vns changed to -Vps

SHA-1: 18526a47fd3a2cbaced57e07aa235dbeea2c4e59

beutlich, 2017-04-25:
Please also add an descriptive comment to package OpAmps itself. Thanks.

kristinmajetta, 2017-07-06:
comment strings added to parameter in Modelica.Electrical.Analog.Examples.OpAmps models
SHA-1: 2a48d4f489e6a0de35c631ae4c615825f047f2a5

kristinmajetta, 2017-07-06:
descriptive string for package OpAmps added
SHA-1: c3d2266578d11051d1c90b3a080e93e4b9f691c5

closed 2017-07-06 by kristinmajetta