modelica/ModelicaStandardLibrary #2234: Add ModelicaWarning

beutlich, 2017-04-05:
I'd like to update ModelicaUtilities.h for the new ModelicaWarning functions of [Modelica v3.4](https://trac.modelica.org/Modelica/ticket/1967) already now. How to do it in backwards-compatible way?

```c
#if defined(MODELICA_UTILITIES_VER) && MODELICA_UTILITIES_VER > 0x030400
void ModelicaWarning(const char* msg);
#endif
```

@HansOlsson @sjoelund Please give feedback.

sjoelund, 2017-04-05:
ModelicaUtilities.h is part of the Modelica specification; it should not be part of the MSL sources except perhaps in documentation.

beutlich, 2017-04-05:
Neither is true currently. ModelicaUtilities.h is part of the MSL and not part of the MLS.

HansOlsson, 2017-04-05:
I agree that FMI might have a better solution, and we should have thought more about this.

On one hand ModelicaUtilities.h shouldn't have been part of MSL; since the implementation is provided by the tools. 

But if we want to compile the library sources (for MSL and also for user models) in a tool-independent way - then we need ModelicaUtilities.h, and finally if a tool has their own implementation and want to rebuild the ".lib" files in MSL it makes sense to use their own ModelicaUtilities.h.

The solution seems to be that MSL contains a default ModelicaUtilities.h (in a separate directory - that we standardize the name of) - and it is documented to change that IncludeDirectory to something else. If a tool wants to use ModelicaUtilities.h from MSL they just add e.g. "modelica://Modelica/Resources/Standard" as IncludeDirectory.

-
For the actual question: as far as I recall about C-linking as long as ModelicaWarning isn't called it doesn't matter if it is declared; so the change above seems to work.