modelica/ModelicaStandardLibrary #2236: Modelica.Media.Examples.MoistAir violates min/max attributes

otronarp, 2017-04-12:
In the reference results for `Modelica.Media.Examples.MoistAir` the variable `medium.x_sat` has a value of ~1.6  at the end of the simulation. `x_sat` is a `MassFraction` that has `max=1`, so the reference result clearly violates the max attribute for that variable. Is this a modelling error or is the reference result wrong?

If we we look at `X_sat` (in `Modelica.Media.Air.MoistAir`) it has the following equation:
`X_sat = min(p_steam_sat * k_mair / max(100 * Constants.eps, p - p_steam_sat) * (1 - Xi[Water]), 1.0) "Water content at saturation with respect to actual water content";`
so it's enforced to be less than 1 whereas `x_sat` is not:
`x_sat = k_mair*p_steam_sat/max(100*Constants.eps, p - p_steam_sat);`

HansOlsson, 2017-04-12:
It seems it is wrong to use MassFraction for x_sat - it is steam contents divided by dry air - so kg/kg, but not with max=1. As far as I can tell there is no max-less MassFraction - the most appropriate type seems to be Modelica.SIunits.RelativeDensity; alternative use MassFraction and modify max.

otronarp, 2017-04-12:
I also see the errors bellow so someone might want to take a look at all the MassFractions in `Modelica.Media.Air.ReferenceMoistAir`, `Modelica.Media.Air.MoistAir` and `Modelica.Media.Examples.TwoPhaseWater.ExtendedProperties` and make sure that the choice of MassFraction type and it's min/max attributes makes sense.

`Modelica.Media.Examples.ReferenceAir.MoistAir`:
```
[ERROR]: At time 0.4465, variable medium.X_liquid = -9.37525e-06 is outside its region ([0,1]) as defined by its min and max attributes.
```
`Modelica.Media.Examples.ReferenceAir.MoistAir1`:
```
[ERROR]: At time 0, variable volume.medium.X_liquid = -0.00448846 is outside its region ([0,1]) as defined by its min and max attributes.
[ERROR]: At time 0, variable fixedMassFlowRate.medium.X_liquid = -0.325633 is outside its region ([0,1]) as defined by its min and max attributes.
[ERROR]: At time 0, variable ambient.medium.X_liquid = -0.00448846 is outside its region ([0,1]) as defined by its min and max attributes.
```

`Modelica.Media.Examples.ReferenceAir.MoistAir2`:
```
[ERROR]: At time 0, variable volume.medium.X_liquid = -0.0122993 is outside its region ([0,1]) as defined by its min and max attributes.
[ERROR]: At time 0, variable fixedMassFlowRate.medium.X_liquid = -0.500787 is outside its region ([0,1]) as defined by its min and max attributes.
[ERROR]: At time 0, variable fixedMassFlowRate.medium.x_sat = 1.02342 is outside its region ([0,1]) as defined by its min and max attributes.
[ERROR]: At time 0, variable ambient.medium.X_liquid = -0.0122993 is outside its region ([0,1]) as defined by its min and max attributes.
[ERROR]: At time 0, variable volume1.medium.X_liquid = -0.0122993 is outside its region ([0,1]) as defined by its min and max attributes.
```

`Modelica.Media.Examples.TwoPhaseWater.TestTwoPhaseStates`:
```
[ERROR]: At time 20.08, variable medium.x = 1 is outside its region ([0,1]) as defined by its min and max attributes.
```

wischhusen, 2017-04-13:
There seem to be multiple problems:

1) The tester model Modelica.Media.Examples.ReferenceAir.MoistAir increases X_sat while Xi[water] stays constant. Thus, the liquid fraction X_liquid = Xi[water] - X_sat  turns negative at some time. Therefore, we need a numerical guard in the following equations:
  X_sat = k_mair/(p/p_steam_sat - 1 + k_mair);
  X_liquid = Xi[Water] - X_sat;
as implemented in Modelica.Media.Air.MoistAir.BasePorperties .

p_steam_sat should also be limited in order to have reasonable numbers for x_sat and phi.

2) The absolute water content x (kg H2O / kg dry air) has no upper limit since the mixture of water and air can have free mass contributions. Air could become a tiny mass fraction.

Here as I understand a potential solution could be to modify the max attribute to Modelica.Constants.inf. Is that correct? MassFraction sounds better than RelativeDensity ... Or we define a local new type called AbsoluteWaterContent with the appropriate attributes. 

What do you think?

beutlich, 2018-01-21:
Resolved in master.

closed 2018-01-21 by beutlich