modelica/ModelicaStandardLibrary #2237: %name field missing

ceraolo, 2017-04-12:
In Modelica.Blocks.Logical there are 8 models without a %name field:
GreaterThreshold, GreaterEqualThreshold,  LessThreshold, LessEqualThreshold, Greater, GreaterEqual,  Less, LessEqual.

I think it is a good idea that all MSL models have a name field included the ones listed above. It eases looking at the output variables in tools.

beutlich, 2017-04-12:
Not sure if this will happen, since Blocks.Logical is superseded by Blocks.MathBoolean, see #2164.

dietmarw, 2017-04-12:
@ceraolo has a point here though. I mean even if there is something planned (whenever that should be) `MathBoolean` can not really replace `Logical` and so the library officer (@otter) should make up his mind if there is a real chance to deprecate `Logical` or not. In the further case the some of the blocks from `Logical` need to be implemented in `MathBoolean` (there are still some missing) and in the latter case it should simply be fixed with whatever issues occur. Starting with removing the useless comment that is referred to in #2164.

AHaumer, 2017-05-14:
resolved in master

commit 96f46089 (2017-05-14): fixes #2237 (missing %name)
closed 2017-05-14 by AHaumer