modelica/ModelicaStandardLibrary #2241: External function wrapper dgees should not pass garbage for function pointer

henrikt-ma, 2017-04-19:
The `Matrices.LAPACK.dgees` function passes garbage to the third argument of the external `dgees` function where it expects a function pointer.

Instead, `Matrices.LAPACK.dgees` needs to call a wrapper provided in ModelicaUtilities that is free of the function pointer argument, and in turn calls the external `dgees` with something more sensible.

sjoelund, 2017-04-19:
> If SORT = 'N', SELECT is not referenced.

And MSL sets SORT="N". So the garbage should not matter as long as the input is the same length as a function pointer...

beutlich, 2017-04-19:
Previosly there was such a C interface wrapper `c_inter_dgees`, which was reverted by b4670538d641fb9636f4b4960d6eb6d3bfbe62a4 for #960. Since the third argument is not used in `dgees` (second argument is `N`, not `S`), it works, though it looks like a hack.

henrikt-ma, 2017-04-19:
How do you make it work when the compiler is provided a correct function prototype for type checking the function call?

I'd say it is a hack, and that it doesn't work.

beutlich, 2017-04-19:
Can you post the compiler output please. So far I did not see any issue, neither GCC nor MSVC.

henrikt-ma, 2017-04-21:
It doesn't work if you type check the call of the external function, which we do by means of providing a correct function prototype.  Here is a dummy example showing the principle (we happen to use the f2c implementation, but that shouldn't matter for the current discussion):
```C
#include <stdlib.h>
#include <string.h>

typedef int f2c_integer;
typedef unsigned int f2c_uinteger;
typedef int f2c_logical;
typedef float f2c_real;
typedef double f2c_doublereal;
typedef f2c_logical (*f2c_L_fp)();

f2c_integer dgees_(char* jobvs, char* sort, f2c_L_fp select, f2c_integer* n, f2c_doublereal* a, f2c_integer* lda, f2c_integer* sdim, f2c_doublereal* wr, f2c_doublereal* wi, f2c_doublereal* vs, f2c_integer* ldvs, f2c_doublereal* work, f2c_integer* lwork, f2c_logical* bwork, f2c_integer* info); /* Signature derived from clapack.h. */

int
main()
{
  f2c_doublereal T[1] = { 0.0 };
  f2c_doublereal Z[1] = { 0.0 };
  f2c_doublereal eval_real[1] = { 0.0 };
  f2c_doublereal eval_imag[1] = { 0.0 };
  f2c_doublereal work[1] = { 0.0 };
  f2c_logical bwork[1] = { 0 };
  f2c_integer llwork[1] = { 0.0 };
  f2c_integer info;
  f2c_integer select[1] = { 0 };
  f2c_integer n = 1;
  f2c_integer lda = 1;
  f2c_integer sdim = 1;
  f2c_integer ldvs = 1;
  f2c_integer $tmp20;
  f2c_logical* $tmp21;
  char* jobvs = strdup("V");
  char* sort = strdup("N");

  dgees_(jobvs, sort, select, &n, T, &lda, &sdim, eval_real, eval_imag, Z, &ldvs, work, llwork, bwork, &info);

  free(jobvs);
  free(sort);
  return 0;
}
```

Clang correctly detects and reports the problem:
```
dgees-sample.c:34:23: warning: incompatible pointer types passing 'f2c_integer [1]' to parameter of type 'f2c_L_fp' (aka 'int (*)()') [-Wincompatible-pointer-types]
  dgees_(jobvs, sort, select, &n, T, &lda, &sdim, eval_real, eval_imag, Z, &ldvs, work, llwork, bwork, &info);
                      ^~~~~~
dgees-sample.c:11:54: note: passing argument to parameter 'select' here
f2c_integer dgees_(char* jobvs, char* sort, f2c_L_fp select, f2c_integer* n, f2c_doublereal* a, f2c_integer* lda, f2c_integer* sdim, f2c_doublereal* wr, f2c_doublereal* wi, f2c_doublereal* vs, f2c_integer* ldvs, f2c_doublereal* work, f2c_integer* lwork, f2c_logical...
                                                     ^
```

sjoelund, 2017-04-21:
We can't really create a wrapper in MSL though. The function only depends on the lapack library and ModelicaExternalC does not depend on lapack. And even if we wrote a wrapper in C, the types would be totally wrong since we don't know which fortran implementation to compile against... It would be nice if we could pass NULL() or something in the external declaration though..

beutlich, 2017-04-21:
Yes, we currently have to live with that warning, except we decide do get rid of the pure lapack dependency and switch to a clapack dependency, where sources can be provided by MSL itself (and no longer by the tool-vendors).

Note, there is another unavoidable warning: the use of 'mktemp' is dangerous, better use 'mkstemp' in ModelicaMatIO.c, which has to be accepted as-is. There is no clean solution Right now available.

henrikt-ma, 2017-04-21:
OK, that's a good point, @sjoelund.  Having a way to pass NULL might be the best we can ask for here.  Any good ideas for what that would look like?

beutlich, 2017-04-21:
Create an external object that returns NULL in the ctor.

henrikt-ma, 2017-04-21:
Could you elaborate that a bit?  How should the external object be passed over the FORTRAN 77 interface?

sjoelund, 2017-04-21:
I thought more to write:
```mo
external "FORTRAN 77" dgees(
    "V",
    "N",
    NULL(),
    n,
    T,
```
which would simply pass NULL since that works for either function or data pointers (but would require a Modelica extension). If you pass a data pointer (`void*`/external object), some compilers do give warnings as well (in particular, C++ compilers):
```c
#include <stdlib.h>

int a(int (*b)())
{
  return b();
}

int c()
{
  return a(NULL);
}

int d()
{
  void *e=(void*)0x1;
  return a(e); /* Not allowed in C++ */
}
```

I also don't think sticking all clapack/lapacke sources (~60MB) into MSL would be a good idea (although perhaps it would be OK to depend on it as a system library and create C wrappers in a new ModelicaExternalC thing).

beutlich, 2017-04-21:
60MB? If you only take the relevant clapack files it is 4MB.

beutlich, 2017-04-21:
We really should think of providing a self-contained MSL and do not put the [work load](https://github.com/modelica/Modelica/blob/master/Modelica/Resources/C-Sources/_readme.txt#L49-L52) on the tool vendors.

sjoelund, 2017-04-21:
The problem with packaging half a lapack implementation comes whenever someone wants to use a function not exported by MSL.

beutlich, 2017-04-21:
Hm? Clapack is 1559 files in 18MB. Ignoring the c-, s- and z-file variants, we come up with 349 d-files (4MB).

beutlich, 2017-04-21:
Do you know of any tool that uses the vanilla FORTRAN interface to lapack? To my knowledge all tool vendors map to the C interface to Clapack with arrayLayout="columnMajor".

henrikt-ma, 2017-04-21:
I think it would be worthwhile talking to the tool vendors to see if there is any what would object to providing clapack on the tool side.

beutlich, 2017-04-21:
Let's do so: @HansOlsson @svorkoetter @jespermattsson

HansOlsson, 2017-04-21:
Dassault Systemes already use the clapack-interface as far as I recall, and standardizing on a C-interface would help with this, but we would have to do some minor changes to support a C-interface so would prefer delaying that change a few months. 
Note that 'NULL' has been complicated in C++ (that's why C++11 added std::nullptr); and in general it is safer to use plain 0.
Switching to a C-implementation of Lapack would help with standardizing the String-arguments as well.

beutlich, 2017-04-21:
One reason that speakes against the proposal is maintenance effort and responsibility: At one of the last MAP-Lib meetings it was decided to no longer include any more external third-party C libraries into MSL for the sake of mainetance effort and platform independency. (Tbh, this was not a decision I welcomed.)

Clapack stopped at v3.2.1 but has bugs, but f2c requires F77 and therefore does no longer work for newer lapack versions which are based on F90. (For that reason we backported several critical bug fixes of lapack > v3.2.1 to lapack v3.2.1 and run f2c ourselves, i.e. have a custom v3.2.1.)

henrikt-ma, 2017-04-21:
To make it a tool requirement to provide clapack is _not_ including any more external third-party C in MSL.  It is a language runtime support requirement, and a pretty reasonable one for Modelica given the role of MSL and the use of LAPACK therein.

beutlich, 2017-04-21:
It is not a language support requirement to me, rather it is a library requirement.

henrikt-ma, 2017-04-24:
As @sjoelund noted, passing the wrong type of pointer doesn't merely result in a warning when a tool is generating C++ code.  For such tools, coping with the current hack may come down to something as awkward as generating a null pointer for any Integer array that looks like `dummyFunctionPointerNotUsed`.  I am sure this isn't a situation we want in the long run.

I just created a language ticket for discussion of how to provide language support for dealing with the problem: https://trac.modelica.org/Modelica/ticket/2183