modelica/ModelicaStandardLibrary #2243: Missing initial assignments in Modelica.Fluid.Examples.ControlledTankSystem.ControlledTanks

beutlich, 2017-04-24:
The test model T2243 is a reduced model of Modelica.Fluid.Examples.ControlledTankSystem.ControlledTanks and gives warnings about missing initial assignments

```mo
model T2243
  Modelica.Fluid.Examples.ControlledTankSystem.Utilities.RadioButton start(reset={stop.on,shut.on}, buttonTimeTable={20,280})
    annotation (Placement(transformation(extent={{-66,40},{-46,60}})));
  Modelica.Fluid.Examples.ControlledTankSystem.Utilities.RadioButton stop(reset={start.on,shut.on}, buttonTimeTable={220,650})
    annotation (Placement(transformation(extent={{-66,10},{-46,30}})));
  Modelica.Fluid.Examples.ControlledTankSystem.Utilities.RadioButton shut(reset={start.on,stop.on}, buttonTimeTable={700})
    annotation (Placement(transformation(extent={{-66,-20},{-46,0}})));
  annotation (uses(Modelica(version="3.2.2")), experiment(StopTime=900));
end T2243;
```

The reason is that in [RadioButton](https://github.com/modelica/Modelica/blob/master/Modelica/Fluid/Examples/ControlledTankSystem.mo#L432) only the pre-values of reset are assigned since f090fb3ab4368065c25cdfed845235fe97b12f39.

MartinOtter, 2018-06-05:
I am analysing this model since some time and do not understand why initial conditions are missing. For example, Dymola reports missing initial condition for start.table.y. However, there are the continuous time equations:
````
start.table.y = start.table.combiTable.y >= 0.5
start.table.combiTable.y = <continuous time equation>
````
This means start.table.y is well-defined also during initiatialization. This might be an issue with the tool. Therefore, I have added Hans as further Assignee.

HansOlsson, 2018-06-05:
I have soon made the pull request.

PR #2591 (2018-06-13): Add start value for RadioButton
To make a clean translation for ControlledTanks we need start-values for both table.y and reset.
As far as I understand they are not relevant after the initialization (since we have equations) - and not used during initialization (since when-conditions aren't active unless using when initial). 
However, they should not cause any problems.

Close #2243
closed 2018-06-13 by casella