modelica/ModelicaStandardLibrary #2247: Questions on Modelica.Mechanical.Rotational.Components.ElastoBacklash2

UlfNordstrom, 2017-04-25:
Hi,

We have a customer who has asked for some details on the ElastoBacklash2 implementation that I cannot answer. Can you help explain (the odd numbering below is due to additional questions already answered)?

-------
Q1) tau_d
In both ElastoBacklash and ElastoBacklash2, you can see "tau_d = d*w_rel;"
But in the info layer, it says "if phi_rel - phi_rel0 < b/2 then tau_d = 0;"
Can you explain?

Q4) ElastoBacklash equation

--- ElastoBacklash ---
    tau = if abs(b) <= bEps then
             tau_c + tau_d
          else if phi_diff > bMax then
             smooth(0, noEvent(if tau_c + tau_d <= 0 then 0 else tau_c + min(tau_c, tau_d)))  <-- 
          else if phi_diff < bMin then
             smooth(0, noEvent(if tau_c + tau_d >= 0 then 0 else tau_c + max(tau_c, tau_d)))  <--
          else 0;

--- ElastoBacklash2 ---
    tau = if abs(b) <= bEps then 
             tau_c + tau_d
          else if phi_diff > bMax then 
             (if tau_c + tau_d <= 0 then 0 else tau_c + tau_d)  <--
          else if phi_diff < bMin then 
             (if tau_c + tau_d >= 0 then 0 else tau_c + tau_d)
          else 0;

Is "tau_c + min(tau_c, tau_d)" and "tau_c + max(tau_c, tau_d)" correct?
Please explain the exact difference between the two equations.
Please explain the reason why they differ.

Q6) Difference between ElastoBacklash and ElastoBacklash2
I understand the difference between the two is "tau_c + min(tau_c, tau_d)" and "tau_c + max(tau_c, tau_d)" (shown as Q4)
Is my understanding correct?

-------

Best regards,
Ulf

beutlich, 2017-04-25:
1dd9dde5a5dd19b159d7182327f6d22bc1ba309f introduced ElastoBacklash2 for MSL v3.2.2.

beutlich, 2017-10-14:
Q1 is addressed by #2357. Q4 and Q6 are support questions and may be also raised at [Stack Overflow](https://stackoverflow.com/questions/tagged/modelica).

MartinOtter, 2017-10-15:
Q4: The goal is to guarantee that tau is continuous when entering/leaving the backlash region (for simplicity ignoring smooth(..) below):

When going from the backlash region to phi_diff > bMax we have
tau_c > 0, w_rel > 0, tau_d > 0

- if tau_c > tau_d: min(tau_c,tau_d) = tau_d
  -> tau = tau_c + tau_d

- if tau_c < tau_d: min(tau_c,tau_d) = tau_c
  -> tau = tau_c + tau_c -> at phi_diff = bMax, tau = 0 (since tau_c = 0)
   
When going from phi_diff > bMax into the backlash region we have tau_c > 0, w_rel < 0, tau_d < 0:

-> min(tau_c,tau_d) = tau_d
-> tau = if tau_c + tau_d < 0 then 0 else tau_c + tau_d and therefore tau is continuous and >= 0

This means the limitation of the damping term has only an effect when going from the backlash region to phi_diff > bMax and limits this term by the size of the spring torque.

Similar considerations hold at phi_diff < bMin.

Q6: Yes, your understanding is correct (the difference between the two is that in ElastoBacklash the damping term  is limited when going from the backlash into the elastic region in order that the torque is continuous. In ElastoBacklash2 the damping term is not limited in this case and therefore the torque changes discontinuously when going from the backlash into the elastic region).

closed 2017-10-15 by MartinOtter

beutlich, 2017-10-15:
Q1: Yes, the code is correct. But I believe the customer confusion comes from the different consideration of protected auxiliary variables tau_c and tau_d. In the comment
https://github.com/modelica/Modelica/blob/9d5bc998bbe08a153fd622622400e744704a4257/Modelica/Mechanics/Rotational.mo#L2676-L2692
and later in the documentation
https://github.com/modelica/Modelica/blob/9d5bc998bbe08a153fd622622400e744704a4257/Modelica/Mechanics/Rotational.mo#L2676-L2692
both tau_c and tau_d are considered the same way (and set to zero in backlash). However this is not reflected by the actual equation
https://github.com/modelica/Modelica/blob/9d5bc998bbe08a153fd622622400e744704a4257/Modelica/Mechanics/Rotational.mo#L2699-L2702