modelica/ModelicaStandardLibrary #2248: Questions regarding Fluxtubes, HysteresisModelComparison

chria, 2017-05-02:
Hi,

Simulating the model Modelica.Magnetic.FluxTubes.Examples.Hysteresis.HysteresisModelComparison using JModelica result in that the variables:

TellinenTable.Phi
TellinenSoft.Phi
PreisachEverett.Phi

are chosen as states. These are poor choices for states as their magnitude are around 1e-4 and because of that they are missing nominal values. Would it be possible to include nominal values for these? 

A better choice, in this model, for states would be 

TellinenTable.B
TellinenSoft.B
PreisachEverett.B

is this true in general? If so, would it be possible to set state select on these?

Thanks!

Best
/Christian

beutlich, 2017-06-02:
@ThomasBoedrich Did you had a look at this?

ThomasBoedrich, 2017-06-06:
We did have a look at this ticket, but unfortunately at the moment we don't have capacities to implement and test other state variables. However, selecting flux density instead of magnetic flux as state variable should be possible in principle. 
During model development we've experimented with different state variables and start values, each of them resulting in certain benefits and drawbacks (e. g. numerical problems). The present hysteresis models have been sucessfully tested with different simulation environments prior to inclusion into the Modelica Standard Library, including OpenModelica. However, they have not been tested with JModelica at that time. During the tests we did not encounter problems with small values of the automatically chosen states or with missing start values.

hubertus65, 2018-05-24:
From a numerical point of view, it is very obvious that the flux densities are better choices of states in the library example and will lead to more accurate results in any simulation environment. 

In addition, all of the 3 hystereses models are actually written as states by the modeler (with a der() on the flux density B ), so it would basically be a confirmation of what I see as the intent of the modeler to add stateSelect.prefer. Phi as a variable is structurally equivalent, scaled by a constant, but a worse scale basically always. Alo Dymola gets faster with B as a preferred state in all 3 models.