modelica/ModelicaStandardLibrary #2250: Use of CPL1.0 license in ModelicaStrings.c without proper declaration

modelica-trac-importer, 2017-05-09:
**Reported by andreas.hofmann7 at 2017-05-09T09:12:34+01:00**
Hello everyone,

I currently discovered, figuring out what open source software is used within the Modelica C-Sources (for contribution of FMUs to customers), that there is a cross reference to the CPL1.0 license within "ModelicaStrings.c" starting at line 532.
To my understanding the link to the CPL1.0 license here is not sufficient, since the license (​https://opensource.org/licenses/cpl1.0.php) states that:
```
"When the Program is made available in source code form:

    a) it must be made available under this Agreement; and

    b) a copy of this Agreement must be included with each copy of the Program."
```
Am I wrong here, when I think there should be a license file available in the C-Sources directory with the CPL1.0 license or that the CPL1.0 license should be available in the source code?

About using a compiled version of the ModelicaString.c I am uncertain, since I do not know if the BSD that is used for the "ModelicaStrings.c" complies with the CPL1.0.

Maybe using a newer version of the General Purpose Hash Function Algorithms Library can help here, since it is licensed under MIT license.

Furthermore, I think it would be a good idea to discuss the topic of open source software with the tool vendors, since using any file from the Modelica/Resources/C-Sources directory within a FMU or a compiled executable, requires to provide a license to the individual file (they have a wide range of copyright holders) as the 2-Clause BSD license requires this (​https://opensource.org/licenses/BSD-2-Clause):
```
"2. Redistributions in binary form must reproduce the above copyright notice, this list of conditions and the following disclaimer in the documentation and/or other materials provided with the distribution."
```
Best regards,
Andreas

----

Migrated-From: https://trac.modelica.org/Modelica/ticket/2188

beutlich, 2017-05-09:
CPL1.0 code was added in 61763832d4c6ce46403919c11a5cd5ff396d6735 for inclusion of Noise/[ModelicaRandom.c](https://github.com/DLR-SR/Noise/blob/master/Modelica_Noise%201.0%20Beta.1/Resources/Include/ModelicaRandom.c#L251-L280) in MSL v3.2.2 (#1662).

beutlich, 2017-05-09:
@MartinOtter I have two alternative proposals

- Since Modelica.Utilities.Strings.hashString is only referenced from ModelicaTest, but not the MSL itself, it can be removed. We would lose backward-compatibility though).

- Since uthash (BSD-style and already part of the MSL) also contains have a dozen hash algorithms I propose to use one of them. Should be easy to fix then.

Which one do you prefer?

MartinOtter, 2017-06-13:
I am a bit confused:

In File ModelicaStrings.c line 489 a reference to the Common Public License is given. However, the URL of the author points to http://www.partow.net/programming/hashfunctions/index.html and here it is stated:

General Hash Function License
Free use of the General Hash Functions Algorithm Library available on this site is permitted under the guidelines and in accordance with the MIT License. 

I also searched on this web site, but did not find the word "Common Public License".
So, it seems a simple solution is to just remove the wrong statement to the Common Public License in the ModelicaStrings.c file and replace it with the MIT License. The advantage is that no user program is influenced.

If there are good **technical** reasons to use the newly proposed hash function, then it seems fine to use the new proposal. However, it the only reason is the license, then it seems better to just replace the wrong license statement by the MIT license.

beutlich, 2017-06-13:
Confusing, indeed: I downloaded [GeneralHashFunctions_-_C.zip](http://www.partow.net/downloads/GeneralHashFunctions_-_C.zip) at Oct. 22 2015 and it said CPL. I downloaded it again today and it now says MIT, though the file date of GeneralHashFunctions.c was kept at Dec. 12 1999.

Thus, basically this is a non-issue, I will take care of by removing the confusing license comment.

beutlich, 2017-06-20:
Closed via #2262.

closed 2017-06-20 by beutlich