modelica/ModelicaStandardLibrary #2253: Analysis of third-party licenses of C-Sources

beutlich, 2017-05-23:
| C library (MA) | License |
| :-- | :-- |
| ModelicaExternalC | [BSD-2-Clause](https://opensource.org/licenses/BSD-2-Clause) |
| ModelicaIO | BSD-2-Clause |
| ModelicaMatIO | BSD-2-Clause |
| ModelicaStandardTables | BSD-2-Clause |

| C library (MA) | Library dependency (3rd-party) | License |
| :-- | :-- | :-- |
| | CLAPACK | [BSD-3-Clause](https://opensource.org/licenses/BSD-3-Clause) |
| ModelicaMatIO | matio | BSD-2-Clause |
| ModelicaMatIO | zlib| [zlib](https://opensource.org/licenses/zlib-license.php) |
| ModelicaMatIO | hdf5 | [BSD-style](https://support.hdfgroup.org/ftp/HDF5/releases/COPYING) |
| ModelicaMatIO | c99-snprintf | [Mention copyright and notice in source code distributions](https://github.com/weiss/c99-snprintf/blob/master/snprintf.c#L1-L19) |
| ModelicaStandardTables / ModelicaExternalC | uthash | [Mention copyright, conditions and disclaimer in source code distributions](https://github.com/troydhanson/uthash/blob/master/src/uthash.h#L1-L22)|
| ModelicaExternalC | Kiss FFT | BSD-3-Clause |
| ModelicaExternalC | xorshift+, xorshift* | Public domain |
| ModelicaExternalC | DIRLIB | Public domain |
| ModelicaExternalC | Hash function | [CPL1.0](https://opensource.org/licenses/cpl1.0.php) (see #2250) |

@chrbertsch  @hubertus65 @MartinOtter @mwetter

chrbertsch, 2017-05-24:
For reference here is the related ticket for OSS licenses within FMUs: [https://trac.fmi-standard.org/ticket/417]

beutlich, 2017-08-18:
This issue was for documentational purpose only. There is nothing left to do.

closed 2017-08-18 by beutlich