modelica/ModelicaStandardLibrary #2258: Add utility functions for Windows Registry access?

beutlich, 2017-06-02:
I was thinking to add new (Windows-only) functions getRegistryIntValue, getRegistryStringValue and getRegistryRealValue (as well as the corresponding set functions) to Modelica.Utilities.System, but would like to ask @modelica/map-lib-write for their opinion first. You can simply use the thumbup/thumbdown feedback 🍨 button from the menu. Thanks for your feedback.

HansOlsson, 2017-06-02:
I can understand the need to access such settings, but I don't see that the need is that large and thus I wouldn't add it at the moment as I see two issues that indicate a strategic issue:

- It seems odd to have something so OS-specific, and I am not sure we have the resources to maintain all variants and it is hard to draw the line
- We are sort of reinventing the wheel. Other languages and libraries already have such layers - and thus people need to find yet another way of getting that information, and we need to duplicate documentation etc. IF we could provide a thin auto-generated layer to an existing binding it would be easier to maintain and motivate

beutlich, 2017-06-02:
You are right. I forgot, that we already can use the [Python winreg module](https://docs.python.org/3/library/winreg.html) together with the [Python27 I/O](https://github.com/lbl-srg/modelica-buildings/tree/master/Buildings/Utilities/IO/Python27) wrapper blocks of the [Buildings](https://github.com/lbl-srg/modelica-buildings) library. Thus, there is no need to reinvent and maintain a direct Modelica interface.

closed 2017-06-02 by beutlich