modelica/ModelicaStandardLibrary #2259: What is the purpose of Modelica.Utilities.Files.list?

HansOlsson, 2017-06-02:
A colleague of mine stumbled upon this function and I have no good answer.

Basically the function seems a bit misplaced. You can use it interactively - but I don't see a use inside other functions - since you cannot use the result.

However, the internal routines Modelica.Utilities.Internal.FileSystem.readDirectory and Modelica.Utilities.Internal.FileSystem.stat  seem more useful  and can be used in user-written functions (after #639 ).

We could just document that the internal routines can be used directly, or we could add a list-variant that either returns the list as array of strings or calls a user-provided function for each file in the directory.

beutlich, 2017-06-02:
@MartinOtter I was added by 38b630e8be5ceddfae3b75615917976067b860e8.

beutlich, 2017-06-02:
@MartinOtter  It was added by 38b630e8be5ceddfae3b75615917976067b860e8 for MSL 2.1.

beutlich, 2017-10-15:
The internal functions Modelica.Utilities.Internal.FileSystem.readDirectory and Modelica.Utilities.Internal.FileSystem.stat are public and can be called in user-written functions legally.

HansOlsson, 2017-10-16:
> The internal functions Modelica.Utilities.Internal.FileSystem.readDirectory and Modelica.Utilities.Internal.FileSystem.stat are public and can be called in user-written functions legally.

It's true they can be used directly, but Modelica.Utilities.Internal is described as "Internal components that a user should usually not directly utilize" - which in general seems like a good recommendation, but not in this case.

beutlich, 2019-11-04:
Removing milestone after missing feedback from library officer.