modelica/ModelicaStandardLibrary #2268: Illegal modifiers in Modelica.Electrical.Analog.Examples.HeatingNPN_OrGate

HansOlsson, 2017-06-15:
The model Modelica.Electrical.Analog.Examples.HeatingNPN_OrGate modifies protected variables vbc, ibe, vt_t in T1 and T2; causing a warning.

The start-value modifiers added as part of #799 seem good, so I would assume that the variables should not be protected - I could not see when that was changed.

A check-log from Dymola says:
Warning: Modifier for protected component vbc. This is not legal Modelica.
  File: Modelica/Electrical/Analog/Examples/HeatingNPN_OrGate.mo, line 107
  Context: Modelica.Electrical.Analog.Examples.HeatingNPN_OrGate.T1.vbc
Original declaration of vbc:
  File: Modelica/Electrical/Analog/Semiconductors.mo, line 1050
  Context: Modelica.Electrical.Analog.Semiconductors.HeatingNPN.vbc
Warning: Modifier for protected component ibe. This is not legal Modelica.
  File: Modelica/Electrical/Analog/Examples/HeatingNPN_OrGate.mo, line 107
  Context: Modelica.Electrical.Analog.Examples.HeatingNPN_OrGate.T1.ibe
Original declaration of ibe:
  File: Modelica/Electrical/Analog/Semiconductors.mo, line 1054
  Context: Modelica.Electrical.Analog.Semiconductors.HeatingNPN.ibe
Warning: Modifier for protected component vt_t. This is not legal Modelica.
  File: Modelica/Electrical/Analog/Examples/HeatingNPN_OrGate.mo, line 106
  Context: Modelica.Electrical.Analog.Examples.HeatingNPN_OrGate.T1.vt_t
Original declaration of vt_t:
  File: Modelica/Electrical/Analog/Semiconductors.mo, line 1062
  Context: Modelica.Electrical.Analog.Semiconductors.HeatingNPN.vt_t
Warning: Modifier for protected component vbc. This is not legal Modelica.
  File: Modelica/Electrical/Analog/Examples/HeatingNPN_OrGate.mo, line 127
  Context: Modelica.Electrical.Analog.Examples.HeatingNPN_OrGate.T2.vbc
Original declaration of vbc:
  File: Modelica/Electrical/Analog/Semiconductors.mo, line 1050
  Context: Modelica.Electrical.Analog.Semiconductors.HeatingNPN.vbc
Warning: Modifier for protected component ibe. This is not legal Modelica.
  File: Modelica/Electrical/Analog/Examples/HeatingNPN_OrGate.mo, line 127
  Context: Modelica.Electrical.Analog.Examples.HeatingNPN_OrGate.T2.ibe
Original declaration of ibe:
  File: Modelica/Electrical/Analog/Semiconductors.mo, line 1054
  Context: Modelica.Electrical.Analog.Semiconductors.HeatingNPN.ibe
Warning: Modifier for protected component vt_t. This is not legal Modelica.
  File: Modelica/Electrical/Analog/Examples/HeatingNPN_OrGate.mo, line 126
  Context: Modelica.Electrical.Analog.Examples.HeatingNPN_OrGate.T2.vt_t
Original declaration of vt_t:
  File: Modelica/Electrical/Analog/Semiconductors.mo, line 1062
  Context: Modelica.Electrical.Analog.Semiconductors.HeatingNPN.vt_t

beutlich, 2017-10-05:
ibe and vbc was modified by 3cffe109e717c42c05bb1f69b2780c1dbdc1a523. vt_t was modified by 81ee9b73eff04368d0f5bc76aa065b84cc211e57. All fine. But they were set to proteced recently by 24ced86e8069ca744c38ca21cc6712f748b91ca5, which should be reverted then.

beutlich, 2017-10-05:
Resolved by eafbcbdc2233cd9551edec0a3fe0f623b92d2317.

closed 2017-10-05 by beutlich