modelica/ModelicaStandardLibrary #2269: Add adaptor components to Mechanics.Translational

beutlich, 2017-06-15:
There are TorqueToAngleAdaptor and AngleToTorqueAdaptor in Modelica.Mechanics.Rotational.Components, however the corresponding elements are missing in Modelica.Mechanics.Translational.Components.

beutlich, 2017-09-20:
@Claytex people are creative meanwhile, see there [Tech Blog](https://t.co/gsdpIlqkIB). 👍

AHaumer, 2017-11-29:
Working on this issue.
I'll open a similar ticket for thermal and electric domain.

dietmarw, 2017-11-30:
The added new components are missing any test models. Either in ModelicaTest or as runnable examples so that simulation errors can be checked for.

AHaumer, 2017-11-30:
Sorry Dietmar to correct you: there is a test model (same as in Rotational).
Modelica.Mechanics.Translational.Examples.GenerationOfFMUs
Anyhow: Thanks for reopening this ticket. 
I'm developing a solution based on the same partial adapter for different domains (electric, thermal). 
Therefore I'm going to close #2399 and #2400.

AHaumer, 2017-12-03:
Partly resolved with general solution (Translational, HatTransfer). Remaining: Electrical.Analog

commit 457c3383 (2017-11-29): resolves #2269
commit 49e9cdbf (2017-12-03): partly resolves #2269
commit d2e71677 (2017-12-10): resolves #2269
closed 2017-12-10 by AHaumer