modelica/ModelicaStandardLibrary #2276: Add more examples to FundamentalWave libraries

christiankral, 2017-07-07:
The examples of ``Modelica.Electrical.Machines.Examples`` shall be adopted for 
- ``Modelica.Magnetic.FundamentalWave.Examples``
- ``Modelica.Magnetic.QuasiStatic.FundamentalWave``

christiankral, 2017-07-08:
Relevant examples are added.

christiankral, 2017-09-19:
Re-structuring examples on Diagram layer

christiankral, 2017-09-19:
I am also unifying the documentation format and fix wrong documentation

christiankral, 2017-09-19:
Somehow the function  ``wrapAngle`` and the block ``WrapAngle`` disappeared, even though added by #2050. This broke some models; so I re-integrated the function and block and the machine models now work fine again.

beutlich, 2017-09-19:
> Somehow the function  ``wrapAngle`` and the block ``WrapAngle`` disappeared, even though added by #2050. 

It never was added to maint/3.2.2 branch. Furthermore, I am not sure if it is worth to maintain this branch.

christiankral, 2017-09-19:
Turing quasi static FundamentalWave examples into real polyphase machine examples.

beutlich, 2017-10-09:
The new example Modelica.Magnetic.QuasiStatic.FundamentalWave.Examples.BasicMachines.InductionMachines.IMC_YD currently fails to initialize in SimulationX. Since the model previously was tweaked to run in OMC, I wonder if it has a general issue.

christiankral, 2017-10-09:
The tweak was a tool dependent annotation which is currently removed again in the MSL, see #2332. The option is to use the Newton solver for nonlinear systems, see https://trac.openmodelica.org/OpenModelica/ticket/4538 and https://trac.openmodelica.org/OpenModelica/ticket/4539. 

That's all which has been tweaked.

beutlich, 2017-10-09:
Any chance to provide better start values or use homotopy?

christiankral, 2018-02-05:
Removed ```SMR_Inverter``` example in 7f815ab and 6c44462 since it do not make sense to demonstrate quasi static behavior of a transient example case. This is a backwards compatible change, since the models are not included in the official release of MSL 3.2.2.

christiankral, 2018-05-15:
Removed obsolete model in fa63e56 and d12767f. Implementation of new models is now completed.

closed 2018-05-15 by christiankral