modelica/ModelicaStandardLibrary #2279: Unify communiciation intervals of electric machine models

christiankral, 2017-07-18:
The communication interval of the electric machine models are set differently in the Machines and FundamentalWave examples. All communication intervals shall be unified to ``experiment(Interval=1E-4,...)`` in order to 
- more accurately show transient torque characteristics 
- better compare results across Machines and FundamentalWave examples

christiankral, 2017-07-18:
@GallLeo I am not sure what impact this change on regression tests has. It may be possible that we have to create a new baseline data set for comparing future MSL versions -- depending on the implementation of the regression test tool (which I am not familiar with).

christiankral, 2017-07-18:
The only exceptions are (due to the longer simulation time):
- SMEE_Generator
- SMEE_Generator_MultiPhase 

For these examples ``experiment(Interval=1E-3,...)``

beutlich, 2017-08-05:
@christiankral Is this issue resolved?

closed 2017-08-06 by christiankral