modelica/ModelicaStandardLibrary #2282: Linear interpolation fails in CombiTimeTable for large values of time

Mathadon, 2017-07-19:
When a `CombiTimeTable` is used to interpolate a data stream with large values of time (i.e. unix time stamps in the order of 1.4e9), the linear interpolation fails. I.e. the returned result changes discontinuously:
[image]

The top graph shows the wrong result, for `time > 1.42e9`. Below is the result when I subtracted 1.42e9 from the time column in the source file. The interpolation then does work.

Note that the top result is not equal to the constant segment implementation: the pieces still have a slope, it's just way too small. These results were verified using Dymola 2017FD01 and Dymola 2018.

Mathadon, 2017-07-19:
@beutlich I think this is one for you?:)

tbeu, 2017-07-19:
Can you please provide the test data. Thanks.

Mathadon, 2017-07-20:
Example .mat file:
https://www.dropbox.com/s/b7pyq0vc0d2ymbw/test.mat?dl=0 
I had to trim my source file and I can't retest the resulting data set right now but it should be ok..

tbeu, 2017-08-03:
Cannot reproduce in SimulationX 3.8 with MSL 3.2.2 and equidistant output step size of 10s.

beutlich, 2017-08-18:
I finally did some more tests of model [T2282](https://gist.github.com/beutlich/cafdd5274d651dac01dc6ac7e9753004) and it really looks like a tool issue to me.

**SimulationX 3.8.4:**
[image]

**OMEdit 1.12.0\~dev-333-gaf79c51 / OpenModelica 1.12.0\~dev-960-gb53dc60:**
[image]

**Dymola 2018:**
[image]

**MWorks.Sysplorer 2016 V4.0.2.874:**
[image]

closed 2017-08-18 by beutlich

Mathadon, 2017-08-18:
Wow, that's unexpected. Thanks for looking into this in details!

HansOlsson, 2017-08-23:
As far as I can tell the reason for the result in Dymola is that the results are stored in single precision, which means that also the time-axis is in single precision.
Thus the solution is to change settings to store in double precision, and then the results look good.

I haven't tested MWorks Sysplorer but would assume it is the same issue.

beutlich, 2017-08-23:
Indeed, it also helps to set double precision in MWorks.Sysplorer.

[image]

Mathadon, 2017-08-23:
@annvdm does this fix https://github.com/modelica/Modelica/issues/2283 ?

beutlich, 2017-08-23:
@Mathadon #2283 is not a tool issue. It is a modeling issue.