modelica/ModelicaStandardLibrary #2283: Reading data fails in ComiTimeTable after 1e+07 seconds

annvdm, 2017-07-20:
When A `CombiTimeTable` is used to read a txt-file that contains data for a whole year, the sampling of the data changes once the time becomes larger than 1e+07. 
See screenshot below, where sampling of the data clearly changes at this time, even though the source data (i.e. the txt-file) does not show this change.

[image]

I used:
`Modelica.Blocks.Sources.CombiTimeTable tab(
tableOnFile=true, 
tableName="data",
fileName="test.txt",
columns=2:2);`

Data file:
https://www.dropbox.com/s/cpsh1t4uvyzx4nc/test.txt?dl=0

tbeu, 2017-07-24:
Duplicate of #2282?

Mathadon, 2017-07-25:
It possibly has the same cause, but the 'symptom' is different: the resolution seems wrong, not the interpolation type?

tbeu, 2017-08-03:
The data in test.txt is stored with only three significant digits. The problem gets apparent first on lines 171 and 172, which read
```
1.01E+05 ...
1.01E+05 ...
```
but rather should be of higher precision
```
1.006E+05 ...
1.012E+05 ...
```
.

closed 2017-08-03 by tbeu